dbisu/pico-ducky #1: Enter does not work

fokenhelpmy, 2021-06-11:
I have played with this for a couple of hours, seems that when it gets to the ENTER command in the script, it just stops (it does gui r and types but never hits enter)
i figured it could possibly be my laptop, so i tested it on another PC just to make sure. the same problem persists.

i then tried using circuit_python 7 to see if maybe it was a bug in that particular version of 6.3 but same problem again.

not sure what to do to get it working

below is the ducky script i was using

DELAY 3000
GUI r
DELAY 200
STRING https://www.youtube.com/watch?v=dQw4w9WgXcQ
DELAY 2000
ENTER
DELAY 3000
STRING f

dbisu, 2021-06-11:
I was not able to reproduce the issue.  I used a couple different computers and didn't see this issue.
I've tried a couple of different scripts with the ENTER command and all seem to be working.
Looks like I am on CircuitPython 6.2.0-beta.1. I'll try updating to 6.3 over the weekend.

fokenhelpmy, 2021-06-12:
So i nuked the previous install with flash_nuke.uf2.

I then added CircuitPython 6.2.0-beta.1
and adafruit-circuitpython-bundle-6.x-mpy-20210611
then added your original upload from github
and the problem is still there.
Even thou I'm based in South Africa, the keyboard lay-out is still US international.

So not quite sure why its not working (All the other commands seem to work just fine)

dbisu, 2021-06-12:
If you connect up minicom or another terminal program to the board, do you see code 40 for ENTER?
ENTER

[40]
ENTER

[40]

fokenhelpmy, 2021-06-14:
[image]

so checking it on a putty connection, this is what i get, will install minicom on a linux vm as soon as i can.(third world internet problems :) )

fokenhelpmy, 2021-06-14:
So this is strange,

Using the keyshow command in linux, i can see an enter when its at the top of the script or the only line in the script.
It seems to stop only after it follows a string command.(almost like its expecting an end to the string.).

so this is the test script.

DELAY 600
ENTER
STRING v
ENTER

This is the output from showkey on ubuntu

28 is the enter key, and 47 is the v (it never hits the trailing enter)

[image]

dbisu, 2021-06-14:
I just setup a new Pico board with the 6.3.0 firmware. 
This is what I see in showkey using your script from above.
keycode  28 press
keycode  28 release
keycode  47 press
keycode  47 release
keycode  28 press
keycode  28 release
keycode  28 press
keycode  28 release

fokenhelpmy, 2021-06-15:
It seems that its an encoding issue.
I redid the script in nano on ubuntu, and it works like a charm.
I previously did it on notepadd++ on my windows machine(this seems to be the problem, possibly the /n issue).

Thanks for this awesome project and sharing it with us and thanks for being so responsive with my issue.

dbisu, 2021-06-15:
I can reproduce this issue if I convert to windows line endings.
I will see if I can find a fix that works with both line ending types.
If I can't find a fix, I will update the documentation to mentions this problem.

dbisu, 2021-06-15:
I figured out the problem.  The split string still includes the \r character.  The keyboard layout doesn't like that character and the script was actually crashing.
I have a fix on the way.
Thank you for reporting this issue and providing the details on the \r\n line endings.

closed 2021-06-15 by dbisu