dbisu/pico-ducky #5: Problem with backslash

jr551, 2021-08-24:
Is this because I have a UK keyboard layout?

Here is the input:

`DEFAULTDELAY 500
DELAY 500
STRING %USERPROFILE%\\desktop\\test
ENTER
STRING %USERPROFILE%\\desktop\\test2
ENTER
STRING \
ENTER
`

This is what it types:
`

%USERPROFILE%##desktop##test
%USERPROFILE%##desktop##test2
#
`

It just places a hash symbol

dbisu, 2021-08-25:
It is probably because the computer is expecting a UK keyboard layout.  If you have a Linux computer available for testing, could you run this command while plugging in the ducky?
sudo showkeys

I haven't tracked down the UK keycodes yet, so not sure the mapping from US to UK.

peterfritz, 2021-08-25:
Wouldn't it be possible to use a different approach like the one proposed by @elkentaro in Bash Bunny?

He made a script that changes the keyboard language to en-US before running the payload and returns to the default after (as seen [here](https://github.com/hak5/bashbunny-payloads/blob/master/payloads/extensions/setkb.sh)).

Sure, it would make the code execution slower, but all the payloads would be internationalized out of the box, making the project easier to configure, use, and, as a bonus, save some precious space on the Pico by not having to store in it a bunch of language files.

While writing this, I see some problems with this approach, like losing the focus on the currently open application before and after the payload execution and making it only compatible with Windows, probably messing things up in other operating systems. But anyway, it would be nice to have something like this even if disabled by default.

jr551, 2021-08-25:
I do not believe you would need to incooperate all the keyboard layouts, it would be appropriate to just inherit the US keyboard and overwrite just the symbol keys where appropriate, maybe the user can set the keyboard type at the start of the script/config file?
Changing the user's keyboard layout with powershell seems a bit messy.

I can get it to do a backslash using the following (would only work on Windows as far as I'm aware but it would be keyboard independent): Maybe we can incoooperate this for now somehow?

def doBackSlashWin():
    kbd.release_all()
    NUMPAD_9 = 0x61
    NUMPAD_2 = 0x5A
    kbd.press(Keycode.ALT)
    time.sleep(float(defaultDelay)/1000)
    kbd.press(NUMPAD_9)
    time.sleep(float(defaultDelay)/1000)
    kbd.release(NUMPAD_9)
    time.sleep(float(defaultDelay)/1000)
    kbd.press(NUMPAD_2)
    time.sleep(float(defaultDelay)/1000)
    kbd.release_all()

peterfritz, 2021-08-25:
The issue is that officially, the Adafruit CircuitPython HID module only support [en-US](https://github.com/adafruit/Adafruit_CircuitPython_HID/blob/main/adafruit_hid/keyboard_layout_us.py).

The setup process is already quite weird. At that point, maybe it would be easier to create some sort of compiler for ducky scripts, this was what Hak5 did on the original USB Rubber Ducky and is way more intuitive and easier to maintain than having that many steps just to set it up.

dbisu, 2021-09-11:
I believe this is a keyboard layout issue.  I'm going to close this issue, as keyboard layout updates are being tracked in Issue #10

closed 2021-09-11 by dbisu