dbisu/pico-ducky #7: Function keys

IamPiter1, 2021-08-25:
I have modified code a little bit, adding function keys, but only F1-F9 are working, F10, 11 and 12 are recognized as F1, Please help someone <3 

`import usb_hid
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keyboard_layout_us import KeyboardLayoutUS
from adafruit_hid.keycode import Keycode
import time
import digitalio
from board import *

duckyCommands = ["WINDOWS", "GUI", "APP", "MENU", "SHIFT", "ALT", "CONTROL", "CTRL", "DOWNARROW", "DOWN",
"LEFTARROW", "LEFT", "RIGHTARROW", "RIGHT", "UPARROW", "UP", "BREAK", "PAUSE", "CAPSLOCK", "DELETE", "END",
"ESC", "ESCAPE", "HOME", "INSERT", "NUMLOCK", "PAGEUP", "PAGEDOWN", "PRINTSCREEN", "SCROLLLOCK", "SPACE",
"TAB", "ENTER", " a", " b", " c", " d", " e", " f", " g", " h", " i", " j", " k", " l", " m", " n", " o", " p", " q", " r", " s", " t",
" u", " v", " w", " x", " y", " z", " A", " B", " C", " D", " E", " F", " G", " H", " I", " J", " K", " L", " M", " N", " O", " P",
" Q", " R", " S", " T", " U", " V", " W", " X", " Y", " Z", "F1", "F2", "F3", "F4", "F5", "F6", "F7", "F8", "F9", "F10", "F11", "F12"]

keycodeCommands = [Keycode.WINDOWS, Keycode.GUI, Keycode.APPLICATION, Keycode.APPLICATION, Keycode.SHIFT, Keycode.ALT, Keycode.CONTROL,
Keycode.CONTROL, Keycode.DOWN_ARROW, Keycode.DOWN_ARROW ,Keycode.LEFT_ARROW, Keycode.LEFT_ARROW, Keycode.RIGHT_ARROW, Keycode.RIGHT_ARROW,
Keycode.UP_ARROW, Keycode.UP_ARROW, Keycode.PAUSE, Keycode.PAUSE, Keycode.CAPS_LOCK, Keycode.DELETE, Keycode.END, Keycode.ESCAPE,
Keycode.ESCAPE, Keycode.HOME, Keycode.INSERT, Keycode.KEYPAD_NUMLOCK, Keycode.PAGE_UP, Keycode.PAGE_DOWN, Keycode.PRINT_SCREEN,
Keycode.SCROLL_LOCK, Keycode.SPACE, Keycode.TAB, Keycode.ENTER, Keycode.A, Keycode.B, Keycode.C, Keycode.D, Keycode.E, Keycode.F, Keycode.G,
Keycode.H, Keycode.I, Keycode.J, Keycode.K, Keycode.L, Keycode.M, Keycode.N, Keycode.O, Keycode.P, Keycode.Q, Keycode.R, Keycode.S, Keycode.T,
Keycode.U, Keycode.V, Keycode.W, Keycode.X, Keycode.Y, Keycode.Z, Keycode.A, Keycode.B, Keycode.C, Keycode.D, Keycode.E, Keycode.F,
Keycode.G, Keycode.H, Keycode.I, Keycode.J, Keycode.K, Keycode.L, Keycode.M, Keycode.N, Keycode.O, Keycode.P,
Keycode.Q, Keycode.R, Keycode.S, Keycode.T, Keycode.U, Keycode.V, Keycode.W, Keycode.X, Keycode.Y, Keycode.Z, Keycode.F1, Keycode.F2,  Keycode.F3,
Keycode.F4, Keycode.F5, Keycode.F6, Keycode.F7, Keycode.F8, Keycode.F9, Keycode.F10, Keycode.F11, Keycode.F12]

def convertLine(line):
    newline = []
    print(line)
    for j in range(len(keycodeCommands)):
			if line.find(duckyCommands[j]) != -1:
				newline.append(keycodeCommands[j])
    print(newline)
    return newline

def runScriptLine(line):
    for k in line:
        kbd.press(k)
    kbd.release_all()
    return

def sendString(line):
    layout.write(line)
    return

def parseLine(line):
    if(line[0:3] == "REM"):
        #comments - ignore
        print("")
    elif(line[0:5] == "DELAY"):
        time.sleep(float(line[6:])/1000)
    elif(line[0:6] == "STRING"):
        sendString(line[7:])
    elif(line[0:13] == "DEFAULT_DELAY"):
        defaultDelay = int(line[14:]) * 10
    elif(line[0:12] == "DEFAULTDELAY"):
        defaultDelay = int(line[13:]) * 10
    else:
        newScriptLine = convertLine(line)
        runScriptLine(newScriptLine)
    return

kbd = Keyboard(usb_hid.devices)
layout = KeyboardLayoutUS(kbd)

#sleep a the start to allow the device to be recognized by the host computer
time.sleep(.5)

# check GPIO0 for program switch
# easiest way to implement is to run a jumper from pin 0 (GPIO0) to pin3 (GND)
progStatus = False
progStatusPin = digitalio.DigitalInOut(GP0)
progStatusPin.switch_to_input(pull=digitalio.Pull.UP)
progStatus = progStatusPin.value

if(progStatus == True):
    #not in programming state, run script file
    defaultDelay = 0
    duckyScriptPath = "payload.dd"
    f = open(duckyScriptPath,"r",encoding='utf-8')
    print("Running payload.dd")
    previousLine = ""
    duckyScript = f.readlines()
    for line in duckyScript:
        line = line.rstrip()
        if(line[0:6] == "REPEAT"):
            for i in range(int(line[7:])):
                #repeat the last command
                parseLine(previousLine)
                time.sleep(float(defaultDelay)/1000)
        else:
            parseLine(line)
            previousLine = line
        time.sleep(float(defaultDelay)/1000)

    print("Done...")
else:
    print("Update new payload file")
`

thatITfox, 2021-08-28:
will this be added to the main code later in the future?

dbisu, 2021-08-28:
It is in the backlog.  Not sure on a timeline right now.

thatITfox, 2021-08-28:
alr 👍

thatITfox, 2021-08-29:
will it be a good idea to put the f10, f11, and f12 in the if else statement, as that's a non permanent solution i found for the f10, f11, and f12 also activating the f1 key

Neradoc, 2021-08-29:
Are there particular intentional limits to ducky script, or would it be ok to extend it to recognize any key in the Keycode module ? You'd still need to convert some codes that are not identical to the ones in Keycode, but you could do something like this, which would support F1-F24 among other things.
```py
duckyConvert = {
    "APP": "APPLICATION",
    "MENU": "APPLICATION",
    "CTRL": "CONTROL",
    "DOWNARROW": "DOWN_ARROW",
    "DOWN": "DOWN_ARROW",
    "LEFTARROW": "LEFT_ARROW",
    "LEFT": "LEFT_ARROW",
    "RIGHTARROW": "RIGHT_ARROW",
    "RIGHT": "RIGHT_ARROW",
    "UPARROW": "UP_ARROW",
    "UP": "UP_ARROW",
    "BREAK": "PAUSE",
    "CAPSLOCK": "CAPS_LOCK",
    "ESC": "ESCAPE",
    "NUMLOCK": "KEYPAD_NUMLOCK",
    "PAGEUP": "PAGE_UP",
    "PAGEDOWN": "PAGE_DOWN",
    "PRINTSCREEN": "PRINT_SCREEN",
    "SCROLLLOCK": "SCROLL_LOCK",
}

def convertLine(line):
    newline = []
    print(line)
    for word in line.split(" "):
        type_word = word.upper()
        if type_word in duckyConvert:
            type_word = duckyConvert[type_word]
        if hasattr(Keycode, type_word):  # accepts any valid entry in Keycode.
            newline.append(getattr(Keycode, type_word))
        else:
            print(f"Unknown key: <{type_word}>")
    print(newline)
    return newline
```

closed 2021-09-04 by dbisu