dbisu/pico-ducky #18: Files Are Only Loaded When Programming Switch.

GamehunterKaan, 2021-09-17:
When Pico Is Plugged It Automatically Launches File Explorer And Interrupts The Payload.To Prevent That Files Should Be Only Loaded When The Board Is In Programming Mode.

dbisu, 2021-09-18:
Potentially a call to storage.disable_usb_drive() could be added to boot.py.  I might make that a second switch, in case someone wants to use the storage during a script for payload distribution or data dumping.

dbisu, 2021-09-18:
Just realized that call is only available in the latest CircuitPython, not CircuitPython 6.3.  I would need to test this project against CircuitPython 7.x before being able to turn off the host PC storage.

GamehunterKaan, 2021-09-19:
I tried running the code with CircuitPython 7.x but for some reason it goes into read only mode and doesn't work.

GamehunterKaan, 2021-09-19:
Ok I Nuked It And Installed Hid Libraries For CircuitPython 7.x.Seems To Be Working Fine For Now.

dbisu, 2021-09-19:
OK.  I'll look at updating the project once 7.x is officially released.  In the meantime, if you want to try with a switch, you can create a boot.py like this.
````
from board import *
import digitalio
import storage

noStorageStatus = False
noStoragePin = digitalio.DigitalInOut(GP15)
noStoragePin.switch_to_input(pull=digitalio.Pull.UP)
noStorageStatus = not noStoragePin.value

if(noStorageStatus == True):
    # don't show USB drive to host PC
    storage.disable_usb_drive()
    print("Disabling USB drive")
else:
    # normal boot
    print("USB drive enabled")
```

GamehunterKaan, 2021-09-19:
I Dont Have Any Jumper Wires Right Now.I Will Give Feedback When They Arrive.(In 2-3 Days Hopefully)

prajwal-bm, 2021-09-21:
this is  not working for me can any one help, i fallowed all the steps but it is not working

GamehunterKaan, 2021-09-21:
It Just Tested It Works Perfectly Fine For Me.Nuke Your Pico First To Reset It,Download CircuitPhyton 7.x,Download Adafruit HID Libraries For 7.x And Then Make Sure You Connect The Corrext Pins. https://i.imgur.com/f28d2u4.jpg

GamehunterKaan, 2021-09-21:
@dbisu [CircuitPython 7.0.0](https://circuitpython.org/board/raspberry_pi_pico/) Is Officially Released.

dbisu, 2021-09-21:
I'll get a card added to update the project to version to 7.0.0.  Sounds like it will be mostly validation testing.

Decapitater, 2021-10-01:
Any updates gentelmen ? The project seems to be updated

dbisu, 2021-10-01:
I might have some time this weekend to look at this again.

closed 2021-10-03 by dbisu