dbisu/pico-ducky #23: thank you for making this it has been amazingly helpful for me but I have a couple changes I have made to the project and I was wondering how I could share my code without making a whole new project ,thank you

acedaawsome, 2021-09-24:
`#this is an edit of https://github.com/dbisu/pico-ducky this edit has added the ability to chose between diffrent payload files without changing the files on the pi and it has changed it to where insted of puting a jumper to stop the code from runing a jumper tells the code to run
#get keyboard info
import usb_hid
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keyboard_layout_us import KeyboardLayoutUS
from adafruit_hid.keycode import Keycode
import time
import digitalio
from board import *
#set commands to be used in payload .dd
duckyCommands = ["WINDOWS", "GUI", "APP", "MENU", "SHIFT", "ALT", "CONTROL", "CTRL", "DOWNARROW", "DOWN",
"LEFTARROW", "LEFT", "RIGHTARROW", "RIGHT", "UPARROW", "UP", "BREAK", "PAUSE", "CAPSLOCK", "DELETE", "END",
"ESC", "ESCAPE", "HOME", "INSERT", "NUMLOCK", "PAGEUP", "PAGEDOWN", "PRINTSCREEN", "SCROLLLOCK", "SPACE",
"TAB", "ENTER", " a", " b", " c", " d", " e", " f", " g", " h", " i", " j", " k", " l", " m", " n", " o", " p", " q", " r", " s", " t",
" u", " v", " w", " x", " y", " z", " A", " B", " C", " D", " E", " F", " G", " H", " I", " J", " K", " L", " M", " N", " O", " P",
" Q", " R", " S", " T", " U", " V", " W", " X", " Y", " Z", "F1", "F2", "F3", "F4", "F5", "F6", "F7", "F8", "F9", "F10", "F11", "F12"]

keycodeCommands = [Keycode.WINDOWS, Keycode.GUI, Keycode.APPLICATION, Keycode.APPLICATION, Keycode.SHIFT, Keycode.ALT, Keycode.CONTROL,
Keycode.CONTROL, Keycode.DOWN_ARROW, Keycode.DOWN_ARROW ,Keycode.LEFT_ARROW, Keycode.LEFT_ARROW, Keycode.RIGHT_ARROW, Keycode.RIGHT_ARROW,
Keycode.UP_ARROW, Keycode.UP_ARROW, Keycode.PAUSE, Keycode.PAUSE, Keycode.CAPS_LOCK, Keycode.DELETE, Keycode.END, Keycode.ESCAPE,
Keycode.ESCAPE, Keycode.HOME, Keycode.INSERT, Keycode.KEYPAD_NUMLOCK, Keycode.PAGE_UP, Keycode.PAGE_DOWN, Keycode.PRINT_SCREEN,
Keycode.SCROLL_LOCK, Keycode.SPACE, Keycode.TAB, Keycode.ENTER, Keycode.A, Keycode.B, Keycode.C, Keycode.D, Keycode.E, Keycode.F, Keycode.G,
Keycode.H, Keycode.I, Keycode.J, Keycode.K, Keycode.L, Keycode.M, Keycode.N, Keycode.O, Keycode.P, Keycode.Q, Keycode.R, Keycode.S, Keycode.T,
Keycode.U, Keycode.V, Keycode.W, Keycode.X, Keycode.Y, Keycode.Z, Keycode.A, Keycode.B, Keycode.C, Keycode.D, Keycode.E, Keycode.F,
Keycode.G, Keycode.H, Keycode.I, Keycode.J, Keycode.K, Keycode.L, Keycode.M, Keycode.N, Keycode.O, Keycode.P,
Keycode.Q, Keycode.R, Keycode.S, Keycode.T, Keycode.U, Keycode.V, Keycode.W, Keycode.X, Keycode.Y, Keycode.Z,
Keycode.F1, Keycode.F2, Keycode.F3, Keycode.F4, Keycode.F5, Keycode.F6, Keycode.F7, Keycode.F8, Keycode.F9,
Keycode.F10, Keycode.F11, Keycode.F12]

def convertLine(line):
    newline = []
    print(line)
    for j in range(len(keycodeCommands)):
		    if line.find(duckyCommands[j]) != -1:
		    	newline.append(keycodeCommands[j])
    print(newline)
    return newline

def runScriptLine(line):
    for k in line:
        kbd.press(k)
    kbd.release_all()

def sendString(line):
    layout.write(line)

def parseLine(line):
    if(line[0:3] == "REM"):
        # ignore ducky script comments
        pass
    elif(line[0:5] == "DELAY"):
        time.sleep(float(line[6:])/1000)
    elif(line[0:6] == "STRING"):
        sendString(line[7:])
    elif(line[0:13] == "DEFAULT_DELAY"):
        defaultDelay = int(line[14:]) * 10
    elif(line[0:12] == "DEFAULTDELAY"):
        defaultDelay = int(line[13:]) * 10
    else:
        newScriptLine = convertLine(line)
        runScriptLine(newScriptLine)

kbd = Keyboard(usb_hid.devices)
layout = KeyboardLayoutUS(kbd)

# sleep at the start to allow the device to be recognized by the host computer
time.sleep(.5)

#check if gp0 is connected to ground
progStatus = False
progStatusPin = digitalio.DigitalInOut(GP0)
progStatusPin.switch_to_input(pull=digitalio.Pull.UP)
progStatus = not progStatusPin.value
defaultDelay = 0
if(progStatus == True):
    #if gp0 is connected to ground run 0payload.dd
    duckyScriptPath = "0payload.dd"
    f = open(duckyScriptPath,"r",encoding='utf-8')
    print("Running payload.dd")
    previousLine = ""
    duckyScript = f.readlines()
    for line in duckyScript:
        line = line.rstrip()
        if(line[0:6] == "REPEAT"):
            for i in range(int(line[7:])):
                #repeat the last command
                parseLine(previousLine)
                time.sleep(float(defaultDelay)/1000)
        else:
            parseLine(line)
            previousLine = line
        time.sleep(float(defaultDelay)/1000)

    print("Done")
else:
    #check if gp1 is connected to ground
    progStatus = False
    progStatusPin = digitalio.DigitalInOut(GP1)
    progStatusPin.switch_to_input(pull=digitalio.Pull.UP)
    progStatus = not progStatusPin.value
    defaultDelay = 0
    if(progStatus == True):
        #if gp1 is connected to ground run 1payload.dd
        duckyScriptPath = "1payload.dd"
        f = open(duckyScriptPath,"r",encoding='utf-8')
        print("Running payload.dd")
        previousLine = ""
        duckyScript = f.readlines()
        for line in duckyScript:
            line = line.rstrip()
            if(line[0:6] == "REPEAT"):
                for i in range(int(line[7:])):
                    #repeat the last command
                    parseLine(previousLine)
                    time.sleep(float(defaultDelay)/1000)
            else:
                parseLine(line)
                previousLine = line
            time.sleep(float(defaultDelay)/1000)
    
        print("Done")
    else:
        #check if gp2 is connected to ground
        progStatus = False
        progStatusPin = digitalio.DigitalInOut(GP2)
        progStatusPin.switch_to_input(pull=digitalio.Pull.UP)
        progStatus = not progStatusPin.value
        defaultDelay = 0
        if(progStatus == True):
            #if gp2 is connected to ground run 2payload.dd
            duckyScriptPath = "2payload.dd"
            f = open(duckyScriptPath,"r",encoding='utf-8')
            print("Running payload.dd")
            previousLine = ""
            duckyScript = f.readlines()
            for line in duckyScript:
                line = line.rstrip()
                if(line[0:6] == "REPEAT"):
                    for i in range(int(line[7:])):
                        #repeat the last command
                        parseLine(previousLine)
                        time.sleep(float(defaultDelay)/1000)
                else:
                    parseLine(line)
                    previousLine = line
                time.sleep(float(defaultDelay)/1000)
        
            print("Done")
        else:
            #check if gp3 is connected to ground
            progStatus = False
            progStatusPin = digitalio.DigitalInOut(GP3)
            progStatusPin.switch_to_input(pull=digitalio.Pull.UP)
            progStatus = not progStatusPin.value
            defaultDelay = 0
            if(progStatus == True):
                #if gp3 is connected to ground run 3payload.dd
                duckyScriptPath = "3payload.dd"
                f = open(duckyScriptPath,"r",encoding='utf-8')
                print("Running payload.dd")
                previousLine = ""
                duckyScript = f.readlines()
                for line in duckyScript:
                    line = line.rstrip()
                    if(line[0:6] == "REPEAT"):
                        for i in range(int(line[7:])):
                            #repeat the last command
                            parseLine(previousLine)
                            time.sleep(float(defaultDelay)/1000)
                    else:
                        parseLine(line)
                        previousLine = line
                    time.sleep(float(defaultDelay)/1000)
            
                print("Done")
            else:
                print("No jumpers placed")

`

acedaawsome, 2021-09-24:
thank you so much I'm sorry to bother I have not used github previously

andreasbrett, 2021-11-16:
@acedaawsome @GamehunterKaan @pvhil I just implemented this over on my fork at https://github.com/andreasbrett/pico-ducky

There are 6 payloads - each correspond to a bridged pin of GP0, GP1, GP2, GP3, GP4, GP5.

closed 2022-02-25 by dbisu