dbisu/pico-ducky #29: Wifi module support?

navin-hariharan, 2021-10-31:
Something similar to the wifiducky? Would be great to see!

thatITfox, 2021-10-31:
that would be an interesting feature, maybe we can make a branch for it, what do you think @dbisu

navin-hariharan, 2021-10-31:
I'll try this from my side tooo!

dbisu, 2021-10-31:
I've been thinking about adding a ports folder for instructions on how to make this project work on boards other than the Raspberry Pi Pico.  Any board with USB HID support that can run CircuitPython should be able to run this project.
As for adding WiFi, seems possible.  Would need to create a protocol from the WiFi board to the Pico.  I don't currently have any ESP8266 boards.

navin-hariharan, 2021-10-31:
> I've been thinking about adding a ports folder for instructions on how to make this project work on boards other than the Raspberry Pi Pico. Any board with USB HID support that can run CircuitPython [...]

I'll try contributing towards getting an ESP!

peterfritz, 2021-11-01:
Integrating files for the ESP8266 paired with serial, connection and wireless logic into this project would make it unnecessarily complex while also making dbisu work three times as hard just to maintain the functionality, merge PRs and fix inevitable bugs.
Creating a separate project like [WiFiDuck](https://github.com/SpacehuhnTech/WiFiDuck) by @spacehuhn would be more practical because it could benefit from the code and logic already written and maintained by dbisu while having two maintainers, meaning that both projects could be updated independently from each other, so the updates could be more frequent without overloading someone. Then both could link to one another helping more people find and use the one that better suits their needs.

thatITfox, 2021-11-01:
> Integrating files for the ESP8266 paired with serial, connection and wireless logic into this project would make it unnecessarily complex while also making dbisu work three times as hard just to [...]

yeah, that's maybe a better idea

dbisu, 2021-11-10:
If anyone wanted to give it a try, this board might be a good place to start.
https://blog.arduino.cc/2021/08/24/power-of-python-for-arduino-nano-rp2040-connect-and-nano-33-ble/

dbisu, 2021-12-07:
I've created the feather-s2-ducky project to add WiFi support.  It will probably take a few weeks to get the WiFi and web support working.
Closing this issue, will track [here](https://github.com/dbisu/feathers2-duckyl).

closed 2021-12-07 by dbisu

dbisu, 2021-12-27:
First pass of wifi enablement on the FeatherS2 is done.  Still a few bugs to workout, but will run scripts from wifi distance.