dbisu/pico-ducky #31: Keyboard layout specific releases

peterfritz, 2021-10-31:
The out-of-the-box support for international keyboard layouts is amazing. 
Unfortunately as it's specific for each and every user, it relies heavily on them, making the process of writing instructions that are clear and concise extremely hard.
I think it would be greatly improved if there was a release with some of the most popular keyboard layouts, so the README instructions would only have to cover the process of choosing, downloading, unzipping and installing the right file from the releases page into the Pico.
Thank you for your time creating and maintaining the awesome project.

dbisu, 2021-10-31:
This is a great idea.  I was going to create a new release now that the keyboard changes are merged into main.  I can create a few release versions.  It might be a few days before I get that accomplished.

dbisu, 2021-10-31:
OK, I was able to get this suggestion done this afternoon.
Release 1.2 has been made with releases for the languages currently available at [Neradoc/Circuitpython_Keyboard_Layouts](https://github.com/Neradoc/Circuitpython_Keyboard_Layouts)

peterfritz, 2021-10-31:
That's perfect! Thank you so much.

closed 2021-10-31 by peterfritz