dbisu/pico-ducky #43: Potential to change name of USB Device in Windows Device menu?

jmath3912, 2022-01-03:
Not necessarily an issue, but rather an inquiry. Is it possible to rename the pico-ducky (or in my case the nano-rp2040-connect-ducky) so that in the Windows Devices menu, it would show up as a name that I define as opposed to "Arduino Nano RP2040 Connect"? And, if possible, would it be done through a setting in boot.py or would I have to modify the .UF2 file?

dbisu, 2022-01-03:
That might be a better question for the CircuitPython Discord.

jmath3912, 2022-01-04:
Ah that's fair.

Bit of info in case anyone is trying to do the same thing I am (which I doubt anyone is, but it's still somewhat useful info regardless): after a bit of research and unnecessary registry noodling, it seems as if Windows pulls the name of the Pico (or in my case the nano rp2040 connect) from the USB driver that gets downloaded on the initial plug-in of the device, which means that a custom or modified driver will have to be written in order for the board to show up in the windows device menu with a custom name. Not too sure on the process of modifying or writing a custom USB driver, but I'm sure there are resources that would go more in depth

peterfritz, 2022-01-14:
Hi, if you're using windows or plan to attack a Windows PC you can create a [Autorun](https://docs.microsoft.com/en-us/windows/win32/shell/autorun-cmds) file with a [label](https://docs.microsoft.com/en-us/windows/win32/shell/autorun-cmds#label) atribute.

Just create a file named `AUTORUN.INF` (in all caps by convention) and paste the following text into it, changing "LabelText" with the desired name. The name can even contain spaces. Mine's named `A inconspicuous device`, but i don't recommend using more than 11 characters.

```c
[AutoRun]
label=LabelText
```

After saving you could hide the file by making it a system file (again, by convention) running the following command from PowerShell or CMD.

```bat
attrib +s AUTORUN.INF
```

P.S. 
I think it would be awesome if the code contained the Autorun file renaming the device to PicoDucky or even adding a duck [icon](https://docs.microsoft.com/en-us/windows/win32/shell/autorun-cmds#icon).

jmath3912, 2022-01-14:
> Hi, if you're using windows or plan to attack a Windows PC you can create a [Autorun](https://docs.microsoft.com/en-us/windows/win32/shell/autorun-cmds) file with a [...]

I see. Would this go into the CIRCUITPY drive or in the root drive on the Pico?

peterfritz, 2022-01-15:
Aren't both the same? 
The file goes in the root of the pico (your's probably named CIRCUITPY) so Windows can read it while mounting the device, be it a flash drive, CD-ROM, SD card, internal or external HDDs and SSDs, it works with everything (in Windows). It's magical.

By the way, there's one more magical file type in Windows that security researchers and others technical in cyber wizardry should definitely learn or at least know that it exists: [desktop.ini](https://docs.microsoft.com/en-us/windows/win32/shell/how-to-customize-folders-with-desktop-ini).

jmath3912, 2022-01-15:
Oh wait I think I understand what you mean now. I've created an `autorun.inf` file and passed a string for the label parameter, but that changes the name of the physical storage drive itself in File Explorer. While that presents its own use case, that wasn't what I was referring to in my original question. What I'm looking for in particular is a way to rename the pico-ducky in the Devices menu from within the settings on Windows (Settings -> Devices). As it stands right now, my device is currently listed as Arduino Nano RP2040 Connect within the Devices section in Windows settings (I modified the code a bit to suit my board). That specific name is what I am trying to modify. I appreciate your help, however, and will look more into desktop.ini as well in case that provides a solution.

peterfritz, 2022-01-15:
I found [this thread](https://forum.arduino.cc/t/change-the-name-of-arduino-pro-micro-shown-in-the-devices-and-printers/660942) regarding a similar issue, but most suggested changes seem local and/or temporary. But even if you find something that works, the method would have to be translated into a completely different microprocessor then tested with CircuitPython for pico-ducky to work.

If there's a solution, I think it involves the code that loads the `.UF2`, way above CircuitPython but if so I don't see how and why would `nano-rp2040-connect-ducky` have "ducky" in it, where the hell did it came from?

I'm sorry, this really doesn't seem to be the best place for we to discuss this issue, I just feel puzzled, baffled, and challenged.

jmath3912, 2022-01-17:
Ah i see i see. After a bit more research in that thread and various other resources, it looks like i may just have to write my own board firmware or perhaps my own circuitpython `.UF2` for my board in order to customize the name. I'm not too sure on the process for either but I'll be looking into it in the near future to see how it's done. 

> Just out of sheer curiosity, and I'm sorry for asking but what would be the use case for this?
Mainly just for testing purposes and further customization on my end. In practice, it could theoretically be used to rename the board into something inconspicuous like SanDisk 32GB Drive or something along the lines of that, as a means of disguising the devices true nature. I mostly plan on using the device as an automated keyboard to speed up the process of certain tasks or at least reduce typing time, so it'd be nice to rename it to something like `DuckyKB` as an easier way for me to differentiate between boards in case I inevitably plan on getting a second one.

In any case, I think this thread has successfully answered some of my questions or rather given me a path I could follow to answer my questions, so I will be closing this issue.

closed 2022-01-17 by jmath3912