dbisu/pico-ducky #45: It’s not working

trace2468, 2022-01-15:


trace2468, 2022-01-15:
I have done exactly what they said

trace2468, 2022-01-15:
Pls help me

trace2468, 2022-01-15:
It is recognised as circuitpython only not a hid device

peterfritz, 2022-01-16:
You have to give us way more detail so we can know exactly what has gone wrong.
Try installing again but this time using [this method](https://github.com/dbisu/pico-ducky/releases/latest) carefully following the instructions step by step.

dbisu, 2022-02-04:
Closing, no activity.

closed 2022-02-04 by dbisu

trace2468, 2022-02-05:
Thank you sir I haven’t put the file in dd format kept in txt. File format
it’s working now thank you very much sir for the your valuable reply can I
get your phone number sir I wanna learn from you how to write ducky scripts
etc.. thank you sir

SJ-XD, 2022-03-31:
hello there i am willing to help and show you how to do this exactly and teach everything about this if you haven't figured it out yes dm me over on discord 

SJ_XD#8529