dbisu/pico-ducky #53: Pico-ducky is not working

zoktik, 2022-03-07:
I don't know why but my Pi Pico-ducky is not working I followed every step is given in your pico-ducky website after completing all things i copy payload script and paste it as payload.dd but is not working 😞😞  so I ordered New pico but Sam thing is happen sir I don't have Enough money for bye one more so please do something and fixed my problem please ......

dbisu, 2022-03-08:
To help debug this issue, please get the debug serial working.
On Windows, use Putty.  On Linux, use minicom
For Windows debug, use Device Manager to find the COM port (usually COM3, but depends on the host computer). Speed is 115200.
For Linux use: `minicom -b 115200 -o -D /dev/ttyACM0`

Once the debug serial port is connected, capture the messages that show up when the device is booted.

You can message me on Twitter for support if needed.

dbisu, 2022-04-04:
Closing, no activity.
Reopen if you are still having problems.

closed 2022-04-04 by dbisu