dbisu/pico-ducky #57: Device is lagging and Starting again.

prashant0230, 2022-03-13:
I am using the raspberry pi pico as a rubber ducky but i am having a problem with it.
When i Plug the rubber ducky into a device it, its start writing the code and stops at midway for some seconds and start writing from the start again.

dbisu, 2022-03-13:
I wonder if you have a bad USB cable that is giving an intermittent connection to your target computer.

prashant0230, 2022-03-17:
I don't understand, are you saying that i do not have good usb cable?

dbisu, 2022-03-17:
I think that might be the issue. Do you have another USB cable you can try to use?
Also, if you could post a log from the debug serial port when this issue is happening, it would be appreciated.

dbisu, 2022-04-04:
Closing, no activity.
Reopen if you are still having problems.

closed 2022-04-04 by dbisu