dbisu/pico-ducky #58: More keyboard layouts

jbb22, 2022-03-14:
Hi, I can't get it working with the swiss keyboard layout. I can download one but it woun't work. Is there any work around? Thanks

dbisu, 2022-03-14:
What errors or other info can you provide?

Per-severance, 2022-03-17:
you just have to change these lines, replace LANG with GERMAN right?
from keyboard_layout_win_LANG import KeyboardLayout
from keycode_win_LANG import Keycode
how do u know what exactly you have to add? I am a bit lost, I tried GERMAN, DE, GER, ger etc and it does not work...
like this:
from keyboard_layout_win_ger import KeyboardLayout
from keycode_win_ger import Keycode

If you know something, that would be great. I cannot provide errors because i cannot run the programm on my environmant.
Thanks for helping!

dbisu, 2022-03-17:
you language setup should look like this:
from keyboard_layout_win_de import KeyboardLayout
from keycode_win_de import Keycode

You also need to ensure these two files are in the lib folder on the pico.
keyboard_layout_win_de.mpy
keycode_win_de.mpy

jbb22, 2022-03-18:
Well, there are some languages officially provided in a git repository (https://github.com/Neradoc/Circuitpython_Keyboard_Layouts/blob/main/PICODUCKY.md) like the german keyboard layout. But not the Swiss-German version. I tried to download the layout from https://www.neradoc.me/layouts/ but it didn't work. I replaced the code in the code.py file as discussed above. With the provided german layout I got it working but not with the swiss.

dbisu, 2022-03-18:
Try updating the code to have these two line:
from keyboard_layout_win_sg import KeyboardLayout
from keycode_win_sg import Keycode

and copying the files in this zip to the lib folder on the pico.
[sg.zip](https://github.com/dbisu/pico-ducky/files/8305371/sg.zip)

jbb22, 2022-03-21:
I replaced the two lines of code and copied the files into the lib folder. Now when I plug it in nothing happens. It gets recognized as usual but nothing gets executed. Thanks for the help!

dbisu, 2022-03-21:
If you are able to open the debug serial port, are there errors being printed there?

Luca10492, 2022-04-09:
Hi, I changed the lines and inserted the two files, but unfortunately it does not work when I connect the raspberry pi pico, then no payload is executed

dbisu, 2022-04-09:
Can you get printouts from the debug serial port?  That information will be helpful to help resolve this problem.

Luca10492, 2022-04-09:
I have got it all now. Its was a little bit tricky but i got it

dbisu, 2022-04-10:
Glad you got it working.  If you don't mind, could you provide feedback on the steps that were tricky so I can improve the instructions?
Thanks.

Luca10492, 2022-04-10:
SureVon meinem Huawei-Telefon gesendet-------- Ursprüngliche Nachricht --------Von: Dave ***@***.***>Datum: So., 10. Apr. 2022, 04:26An: dbisu/pico-ducky ***@***.***>Cc: Luca10492 ***@***.***>, Comment ***@***.***>Betreff: Re: [dbisu/pico-ducky] More keyboard layouts (Issue #58)
Glad you got it working.  If you don't mind, could you provide feedback on the steps that were tricky so I can improve the instructions?
Thanks.

Luca10492, 2022-04-10:
For the German keyboard layout you have to do until step 6. on this page https://github.com/dbisu/pico-ducky. from the step 7 you have to install visual studios code and install the coding language python3 and after then you need to paste this code;
import usb_hid
from adafruit_hid.keyboard import Keyboard
from keyboard_layout_win_gr import KeyboardLayout
from keycode_win_gr import Keycode
import time
import digitalio
from board import *

duckyCommands = ["WINDOWS", "GUI", "APP", "MENU", "SHIFT", "ALT", "CONTROL", "CTRL", "DOWNARROW", "DOWN",
"LEFTARROW", "LEFT", "RIGHTARROW", "RIGHT", "UPARROW", "UP", "BREAK", "PAUSE", "CAPSLOCK", "DELETE", "END",
"ESC", "ESCAPE", "HOME", "INSERT", "NUMLOCK", "PAGEUP", "PAGEDOWN", "PRINTSCREEN", "SCROLLLOCK", "SPACE",
"TAB", "ENTER", " a", " b", " c", " d", " e", " f", " g", " h", " i", " j", " k", " l", " m", " n", " o", " p", " q", " r", " s", " t",
" u", " v", " w", " x", " y", " z", " A", " B", " C", " D", " E", " F", " G", " H", " I", " J", " K", " L", " M", " N", " O", " P",
" Q", " R", " S", " T", " U", " V", " W", " X", " Y", " Z", "F1", "F2", "F3", "F4", "F5", "F6", "F7", "F8", "F9", "F10", "F11", "F12"]

keycodeCommands = [Keycode.WINDOWS, Keycode.GUI, Keycode.APPLICATION, Keycode.APPLICATION, Keycode.SHIFT, Keycode.ALT, Keycode.CONTROL,
Keycode.CONTROL, Keycode.DOWN_ARROW, Keycode.DOWN_ARROW ,Keycode.LEFT_ARROW, Keycode.LEFT_ARROW, Keycode.RIGHT_ARROW, Keycode.RIGHT_ARROW,
Keycode.UP_ARROW, Keycode.UP_ARROW, Keycode.PAUSE, Keycode.PAUSE, Keycode.CAPS_LOCK, Keycode.DELETE, Keycode.END, Keycode.ESCAPE,
Keycode.ESCAPE, Keycode.HOME, Keycode.INSERT, Keycode.KEYPAD_NUMLOCK, Keycode.PAGE_UP, Keycode.PAGE_DOWN, Keycode.PRINT_SCREEN,
Keycode.SCROLL_LOCK, Keycode.SPACE, Keycode.TAB, Keycode.ENTER, Keycode.A, Keycode.B, Keycode.C, Keycode.D, Keycode.E, Keycode.F, Keycode.G,
Keycode.H, Keycode.I, Keycode.J, Keycode.K, Keycode.L, Keycode.M, Keycode.N, Keycode.O, Keycode.P, Keycode.Q, Keycode.R, Keycode.S, Keycode.T,
Keycode.U, Keycode.V, Keycode.W, Keycode.X, Keycode.Y, Keycode.Z, Keycode.A, Keycode.B, Keycode.C, Keycode.D, Keycode.E, Keycode.F,
Keycode.G, Keycode.H, Keycode.I, Keycode.J, Keycode.K, Keycode.L, Keycode.M, Keycode.N, Keycode.O, Keycode.P,
Keycode.Q, Keycode.R, Keycode.S, Keycode.T, Keycode.U, Keycode.V, Keycode.W, Keycode.X, Keycode.Y, Keycode.Z,
Keycode.F1, Keycode.F2, Keycode.F3, Keycode.F4, Keycode.F5, Keycode.F6, Keycode.F7, Keycode.F8, Keycode.F9,
Keycode.F10, Keycode.F11, Keycode.F12]

def convertLine(line):
    newline = []
    print(line)
    for j in range(len(keycodeCommands)):
		    if line.find(duckyCommands[j]) != -1:
		    	newline.append(keycodeCommands[j])
    print(newline)
    return newline

def runScriptLine(line):
    for k in line:
        kbd.press(k)
    kbd.release_all()

def sendString(line):
    layout.write(line)

def parseLine(line):
    if(line[0:3] == "REM"):
        # ignore ducky script comments
        pass
    elif(line[0:5] == "DELAY"):
        time.sleep(float(line[6:])/1000)
    elif(line[0:6] == "STRING"):
        sendString(line[7:])
    elif(line[0:13] == "DEFAULT_DELAY"):
        defaultDelay = int(line[14:]) * 10
    elif(line[0:12] == "DEFAULTDELAY"):
        defaultDelay = int(line[13:]) * 10
    else:
        newScriptLine = convertLine(line)
        runScriptLine(newScriptLine)

kbd = Keyboard(usb_hid.devices)
layout = KeyboardLayout(kbd)

# sleep at the start to allow the device to be recognized by the host computer
time.sleep(.5)

# check GP0 for setup mode
# see setup mode for instructions
progStatus = False
progStatusPin = digitalio.DigitalInOut(GP0)
progStatusPin.switch_to_input(pull=digitalio.Pull.UP)
progStatus = not progStatusPin.value
defaultDelay = 0
if(progStatus == False):
    # not in setup mode, inject the payload
    duckyScriptPath = "payload.dd"
    f = open(duckyScriptPath,"r",encoding='utf-8')
    print("Running payload.dd")
    previousLine = ""
    duckyScript = f.readlines()
    for line in duckyScript:
        line = line.rstrip()
        if(line[0:6] == "REPEAT"):
            for i in range(int(line[7:])):
                #repeat the last command
                parseLine(previousLine)
                time.sleep(float(defaultDelay)/1000)
        else:
            parseLine(line)
            previousLine = line
        time.sleep(float(defaultDelay)/1000)

    print("Done")
else:
    print("Update your payload")

after you paste it you have to save the file code.py and copy the file on your raspberry pi pico.
after then you must download the german keyboard layout: https://www.neradoc.me/layouts/make_zip_bundle.php?url=https://kbdlayout.info/kbdgr&cpversion=

then copy the downloaded gr.mpy to /lib not into /lib/adafruit_hid

I hope you understand my english

Luca10492, 2022-04-10:
import usb_hid
from adafruit_hid.keyboard import Keyboard
from keyboard_layout_win_gr import KeyboardLayout
from keycode_win_gr import Keycode
import time
import digitalio
from board import *

duckyCommands = ["WINDOWS", "GUI", "APP", "MENU", "SHIFT", "ALT", "CONTROL", "CTRL", "DOWNARROW", "DOWN",
"LEFTARROW", "LEFT", "RIGHTARROW", "RIGHT", "UPARROW", "UP", "BREAK", "PAUSE", "CAPSLOCK", "DELETE", "END",
"ESC", "ESCAPE", "HOME", "INSERT", "NUMLOCK", "PAGEUP", "PAGEDOWN", "PRINTSCREEN", "SCROLLLOCK", "SPACE",
"TAB", "ENTER", " a", " b", " c", " d", " e", " f", " g", " h", " i", " j", " k", " l", " m", " n", " o", " p", " q", " r", " s", " t",
" u", " v", " w", " x", " y", " z", " A", " B", " C", " D", " E", " F", " G", " H", " I", " J", " K", " L", " M", " N", " O", " P",
" Q", " R", " S", " T", " U", " V", " W", " X", " Y", " Z", "F1", "F2", "F3", "F4", "F5", "F6", "F7", "F8", "F9", "F10", "F11", "F12"]

keycodeCommands = [Keycode.WINDOWS, Keycode.GUI, Keycode.APPLICATION, Keycode.APPLICATION, Keycode.SHIFT, Keycode.ALT, Keycode.CONTROL,
Keycode.CONTROL, Keycode.DOWN_ARROW, Keycode.DOWN_ARROW ,Keycode.LEFT_ARROW, Keycode.LEFT_ARROW, Keycode.RIGHT_ARROW, Keycode.RIGHT_ARROW,
Keycode.UP_ARROW, Keycode.UP_ARROW, Keycode.PAUSE, Keycode.PAUSE, Keycode.CAPS_LOCK, Keycode.DELETE, Keycode.END, Keycode.ESCAPE,
Keycode.ESCAPE, Keycode.HOME, Keycode.INSERT, Keycode.KEYPAD_NUMLOCK, Keycode.PAGE_UP, Keycode.PAGE_DOWN, Keycode.PRINT_SCREEN,
Keycode.SCROLL_LOCK, Keycode.SPACE, Keycode.TAB, Keycode.ENTER, Keycode.A, Keycode.B, Keycode.C, Keycode.D, Keycode.E, Keycode.F, Keycode.G,
Keycode.H, Keycode.I, Keycode.J, Keycode.K, Keycode.L, Keycode.M, Keycode.N, Keycode.O, Keycode.P, Keycode.Q, Keycode.R, Keycode.S, Keycode.T,
Keycode.U, Keycode.V, Keycode.W, Keycode.X, Keycode.Y, Keycode.Z, Keycode.A, Keycode.B, Keycode.C, Keycode.D, Keycode.E, Keycode.F,
Keycode.G, Keycode.H, Keycode.I, Keycode.J, Keycode.K, Keycode.L, Keycode.M, Keycode.N, Keycode.O, Keycode.P,
Keycode.Q, Keycode.R, Keycode.S, Keycode.T, Keycode.U, Keycode.V, Keycode.W, Keycode.X, Keycode.Y, Keycode.Z,
Keycode.F1, Keycode.F2, Keycode.F3, Keycode.F4, Keycode.F5, Keycode.F6, Keycode.F7, Keycode.F8, Keycode.F9,
Keycode.F10, Keycode.F11, Keycode.F12]

def convertLine(line):
    newline = []
    print(line)
    for j in range(len(keycodeCommands)):
		    if line.find(duckyCommands[j]) != -1:
		    	newline.append(keycodeCommands[j])
    print(newline)
    return newline

def runScriptLine(line):
    for k in line:
        kbd.press(k)
    kbd.release_all()

def sendString(line):
    layout.write(line)

def parseLine(line):
    if(line[0:3] == "REM"):
        # ignore ducky script comments
        pass
    elif(line[0:5] == "DELAY"):
        time.sleep(float(line[6:])/1000)
    elif(line[0:6] == "STRING"):
        sendString(line[7:])
    elif(line[0:13] == "DEFAULT_DELAY"):
        defaultDelay = int(line[14:]) * 10
    elif(line[0:12] == "DEFAULTDELAY"):
        defaultDelay = int(line[13:]) * 10
    else:
        newScriptLine = convertLine(line)
        runScriptLine(newScriptLine)

kbd = Keyboard(usb_hid.devices)
layout = KeyboardLayout(kbd)

# sleep at the start to allow the device to be recognized by the host computer
time.sleep(.5)

# check GP0 for setup mode
# see setup mode for instructions
progStatus = False
progStatusPin = digitalio.DigitalInOut(GP0)
progStatusPin.switch_to_input(pull=digitalio.Pull.UP)
progStatus = not progStatusPin.value
defaultDelay = 0
if(progStatus == False):
    # not in setup mode, inject the payload
    duckyScriptPath = "payload.dd"
    f = open(duckyScriptPath,"r",encoding='utf-8')
    print("Running payload.dd")
    previousLine = ""
    duckyScript = f.readlines()
    for line in duckyScript:
        line = line.rstrip()
        if(line[0:6] == "REPEAT"):
            for i in range(int(line[7:])):
                #repeat the last command
                parseLine(previousLine)
                time.sleep(float(defaultDelay)/1000)
        else:
            parseLine(line)
            previousLine = line
        time.sleep(float(defaultDelay)/1000)

    print("Done")
else:
    print("Update your payload")

Luca10492, 2022-04-10:
[image]

Then copy the downloaded (keyboard_layout_win_gr.mpy & keycode_win_gr.mpy) gr.mpy into /lib (not into /lib/adafruit_hid!!!!!!)

Luca10492, 2022-04-10:
[image]

Luca10492, 2022-04-10:
I belive it will work also with other lanuages.

You must download the Language-Packages for your country.
For this you must go to https://www.neradoc.me/layouts/

### There you must find the url for your Package.
- Go to https://kbdlayout.info/ and selecte this for your country.
- Copy the url and go back to https://www.neradoc.me/layouts/ and paste the url and click "make zip bundle links".
- Unzip the file and copy keyboard_layout_win_LANG.mpy & keycode_win_LANG.mpy (LANG is your language) and paste it on the Raspberry Pi Pico in /lib not in /lib/adafruit_hid and rename it in for example "...win_gr.mpy". 
- After then you must edit the code.py and rename keyboard_layout_win_LANG & keycode_win_LANG it here too. (Without .mpy)

[image]

Luca10492, 2022-04-10:
> you just have to change these lines, replace LANG with GERMAN right?
> from keyboard_layout_win_LANG import KeyboardLayout
> from keycode_win_LANG import Keycode
> how do u know what exactly you have to [...]

I have make a Issues for German keyboard layout

closed 2022-04-10 by dbisu

Luca10492, 2022-04-10:
I hope I could help you