dbisu/pico-ducky #67: No files appear when the .UF2 file gets dropped in Pico

Linkgray, 2022-04-04:
After following the instructions and dropping the .uf2 file into the pico (Also holding down button when connecting), it does reboot into "CIRCUITPY" but there is no files inside but a txt file named "Boot_out". There is no lib folder required for the next steps.
Any details aiding with this will be appreciated.

dbisu, 2022-04-05:
Yes, that sounds correct. You have to create the lib directory on the Pico.

Linkgray, 2022-04-05:
Gotcha. Through the documentation provided and this tutorial I've seen, https://www.youtube.com/watch?v=e_f9p-_JWZw&t=366s from 5:45-6:02 specifically, the steps written made it seem like after plugging in the .uf2 file, files such as .Trashes, Boot_out.txt, code.py and folders like lib and .fseventsd would appear after the self reboot was performed. 

Are the files needed after the "CYRCUITPY" conversion downloadable from a link in your  steps? or do I have to physically create this lib directory?  I only ask because on step 5, it mentions to "copy adafruit.hid to the lib folder in your raspberry." which to me implied it would already be there and not having to create one. 

Moreover, on step 6 it mentions, "save the file as 'code.py' in the root of the Raspberry Pi Pico, overwriting the previous file." But as I mentioned before, there is no previous files to replace as it boots with nothing but "Boot_out."
I feel like I am missing a step. Additional clarify would be greatly appreciated as this seems like a very fun and great tool to have.  
Thanks you!

Linkgray, 2022-04-05:
Disregard! I fixed my issue. I was in fact missing files. I nuked my Pico with the 'Flash_nuke.uf2' file and then re did the installation of your custom .uf2 file and all the files populated!

closed 2022-04-05 by Linkgray

woodsman777007, 2022-04-17:
can you give a link to the 'Flash_nuke.uf2' please

dbisu, 2022-04-17:
From [Learn Adafruit](https://learn.adafruit.com/getting-started-with-raspberry-pi-pico-circuitpython/circuitpython)
https://cdn-learn.adafruit.com/assets/assets/000/099/419/original/flash_nuke.uf2?1613329170