dbisu/pico-ducky #72: pico don't load

WaD45, 2022-04-24:
After following all the steps, i create the payload.dd file and add it to the pico. The pico don't disconnect/reconnect after adding it. Even with setup mode off and USB disabled mode... I tried it on a Win11 computer, WIN10 computer... it just don't inject the payload. It load the usb as a mass storage and if it is in USB Disabled, nothing happens, with differents payloads.

dbisu, 2022-04-25:
I would be helpful if you have a picture of your setup and/or a dump from the debug serial port.

WaD45, 2022-04-25:
A picture of what files are on the pico ? What software can i use for dump ?

dbisu, 2022-04-25:
Data that would be helpful:
Screenshot of the files on the pico
Picture of the pico board, specifically jumper settings
Debug serial data.  On Windows, use PuTTY to connect to the debug serial port, commonly COM3.

dbisu, 2022-05-24:
Try using one of the release bundles for your language.
Closing.

closed 2022-05-24 by dbisu