dbisu/pico-ducky #73: Is there a way to make the pico write faster? + Some other question.

minuspal, 2022-05-07:
Is there anyway theoretically or practically to increase the pico's wpm considering it has a much faster cpu than the original badusb itself?? And will twin duck firmware ever come to the pico ducky? Thanks :)

dbisu, 2022-05-07:
The defaults run the script as fast as the code can parse the script.
There is a plan in the backlog to add some buttons to run the scripts.  It will probably require another switch to set modes, as there is still the need to run the script right away on boot for some use cases.
That will probably be worked on sometime in the next couple of months.

YeetingWaterbottle, 2022-05-22:
Just a random idea but can multi threading make the wpm faster?

dbisu, 2022-05-22:
I would expect the limitation is the HID driver in CircuitPython.
I have not done any benchmarks.  Is there an expected rate vs the actual rate?

YeetingWaterbottle, 2022-05-22:
I actually haven't expiremented with the pico ducky yet, I would love to do some benchmarking after I make one myself.

dbisu, 2022-07-07:
Closing.  I don't think this is a direct problem with this project. You may want to check with the CircuitPython HID project.

closed 2022-07-07 by dbisu