dbisu/pico-ducky #74: Unable to select payloads

Jakubikk, 2022-05-22:
Whenever I ground GPIO4/5/10/11 always payload.dd is executed, no matter the grounded pin.

**To Reproduce**
Steps to reproduce the behavior:

1.Upload 4 files named named payload.dd, payload2.dd, payload3.dd and payload4.dd (each one opens the Mac Spotlight and types its script's number)(included in the zip)
[ducky scripts.zip](https://github.com/dbisu/pico-ducky/files/8750128/ducky.scripts.zip)

2.Ground pin GPIO 5, 10 or 11 (GPIO 10 is grounded in the picture)
[image]

3.Reboot board

4.Spotlight opens, 1 is typed (result visible in screenshot)

**Expected behavior**

Script runs, spotlight opens and "3" is typed (as pin GPIO 10 is grounded in the example)

**Screenshots**

[image]

**Debug info**
Wasn't able to to obtain any :(
**Additional context**

I am using a Mac, that's why I use spotlight, which is summoned with GUI TAB

dbisu, 2022-05-22:
Here is a picture of the schematic I've used to implement the switching behavior.
```
Payload 1 = Pin 6 (GPIO4) to Pin 8 (GND)
Payload 2 = Pin 7 (GPIO5) to Pin 13 (GND)
Payload 3 = Pin 14 (GPIO10) to Pin 8 (GND)
Payload 4 = Pin 15(GPIO11) to Pin 13 (GND)
```
[image]

Jakubikk, 2022-05-23:
That seems to work perfectly, thanks!

closed 2022-05-23 by Jakubikk