dbisu/pico-ducky #76: Tool to flash pico-ducky.

raspberrypi5621, 2022-05-30:
# PyDucky Script

Hello, I made an opensource tool that can flash this awesome tool to
the raspberry pi pico! It flashes circuit python, the adafruit_hid libraries 
and the main code. You can check it [here](https://github.com/raspberrypi5621/pyducky) if you want.

# Running the script

To run the script you just paste these commands to the terminal.
Make sure that you have python3 and git installed. Also, the pico
must be mounted at /media/username/RPI-RP2

`git clone https://github.com/raspberrypi5621/pyducky`
`cd pyducky`
`python3 pyducky.py`

dbisu, 2022-06-07:
This looks really interesting.  I can update the README.md to reference this tool sometime soon.

raspberrypi5621, 2022-06-10:
Thank you very much.

closed 2022-06-12 by dbisu