dbisu/pico-ducky #79: Payload wont run even when everything has been done correctly

Jimby1234, 2022-06-26:
[image]
the image above is what my pico looks like inside. I dont really know the problem so theres nothing really for me to say but what i can say is its not working

dbisu, 2022-06-26:
**Debug info**
 If possible, include debug serial data. On Windows, use PuTTY, connect to the debug serial port (commonly COM3, but could vary)

On Linux, use minicom (minicom -b 115200 -o -D /dev/ttyACM0, where ttyACM0 corresponds to your device)

MonsieurPoutounours, 2022-08-27:
In your picture, the .metadata_never_index file does not have extension shown. So your OS may have "Don't show file extension for known filetypes" checked. 

You payload.dd may indeed be a payload.dd.txt file, which is something that Windows Notepad tend to do if you're not careful (stop using Windows Notepad, use a real file editor). 

Check that.

p-i-c-o, 2022-08-30:
Hi, I have the same issue, Im not sure what i need to look for in general. all files are ok, but payload.dd doesn't run
[image]

dbisu, 2022-08-31:
To have a better idea why it isn't working, I need a dump from the debug serial.
On Linux:
minicom -b 115200 -o -D /dev/ttyACM0

p-i-c-o, 2022-08-31:
How do you do this on MacOS?

dbisu, 2022-08-31:
It looks like you can use screen from a macos terminal
screen /dev/<serial port> 115200
You may need to watch /dev to see what port appears when you plug in the pico.

p-i-c-o, 2022-09-02:
Hey everyone, I'm back,
using Mu Code Editor ( which supports MicroPython ), i found this in the Serial:

[image]

p-i-c-o, 2022-09-02:
Could someone provide code from a working Pico Ducky so that I could compare?

Thanks

dbisu, 2022-09-02:
Main is currently broken.  I have a PR open to fix it and I'll get it merged in tonight or tomorrow.
You could grab the current release [here](https://github.com/dbisu/pico-ducky/releases/tag/v1.3)

dbisu, 2022-09-03:
Main should be fixed now.

N-00-B, 2022-09-04:
[image]

Hi all, I'm relatively new to this stuff so bear with me.. 

I had similar issues with my Pi so I waited until the main was fixed. However, when I followed all steps correctly, even attempting it multiple times, my payloads were not executing. So I downloaded PuTTY and got the debug serial data and this traceback message..

dbisu, 2022-09-04:
Based on the errors, it is likely you missed the update to the README.md instructions that included this line:
Copy adafruit_debouncer.mpy and adafruit_ticks.mpy to the lib folder in your Raspberry Pi Pico.

Copy those two files from the adafruit library bundle (same place that adafruit_hid is in) to the lib folder on the pico.

N-00-B, 2022-09-04:
Oh wow, I did not take note of that at first XD.. Thank you!

dbisu, 2022-09-18:
Closing, main should be working now.

closed 2022-09-18 by dbisu