dbisu/pico-ducky #80: Pico W out!

Sussy-OS, 2022-06-30:
The Pico W (W for WiFI) is now able to be bought. If you can intergerate this, we could wirelessly control our pico, without having to update the payload manualy, you can do it with a GUI right when it's plugged in! (Simalar to the [O.M.G. Plug](https://hak5.org/products/omg-plug) ) This would be *huge* for the Pico-Ducky community, if we could get it to work. Let's see if it works?

dbisu, 2022-06-30:
I saw the announcement on this new board.  I'm not sure how quickly I'll be able to get some boards.  Once I do, I'll work on a port to the Pico W.
For remote support, I'll probably start with some of the things I put in the [Feather-S2 ducky](https://github.com/dbisu/feathers2-ducky) project.

Sussy-OS, 2022-07-01:
> I saw the announcement on this new board. I'm not sure how quickly I'll be able to get some boards. Once I do, I'll work on a port to the Pico W. For remote support, I'll probably start with some of the things I put in the [Feather-S2 ducky](https://github.com/dbisu/feathers2-ducky) project.

Okay. If I get a Pico W, I could volenteer for tester to help? The only problem I am seeing is how to host the GUI/server for remote acsess. If you already have something simalar, then I think that this could be something really great for all Ducky kind. A $6 USB Rubber ducky with WiFi remote acsess. Now we don't need to pay $60! Thanks for reading this, and have a good rest of your day. :)

linuzzx, 2022-07-05:
> The only problem I am seeing is how to host the GUI/server for remote acsess.
  
Can't we get a soft access point running on the pico? So the pico works like a hotspot

dbisu, 2022-07-05:
Yes, that is the plan.  You can see a potential way of doing this in the [Feather-S2 ducky](https://github.com/dbisu/feathers2-ducky) project.  Look at webapp.py. I'll probably clean that up a little once I can get some Pico W boards.

d3k4-k3rb3r0s, 2022-07-06:
https://github.com/raspberrypi/pico-examples

they have some access point examples for pico W if you scroll down.

ratarchivos, 2022-07-24:
did u do it?
how can i download?

dbisu, 2022-07-24:
I don't have any Pico W boards yet.  I expect it will be a couple of months before I am able to get a board.

ratarchivos, 2022-07-25:
i have one ill try to do a local host with a html file that have buttons and a while bucle that ask if is tru and run the code

ratarchivos, 2022-07-25:
i can do that in code.py?

ratarchivos, 2022-07-25:
and i can put a button for change usb drivr on or off?

dbisu, 2022-07-25:
> and i can put a button for change usb drivr on or off?

No, previous experience shows that you can only change the state of the usb drive on boot. I tried doing this in the featherS2-ducky and determined there wasn't a way to make it work.

chriskalv, 2022-07-29:
Using the Pico W as some kind of much much cheaper WiFi Duck device (check [this guide](https://null-byte.wonderhowto.com/how-to/hack-computers-over-wi-fi-with-wifi-duck-payload-deliverer-0296285/) out), would be absolutely **phenomenal**.

dbisu, 2022-09-15:
I'm still working on getting a Pico W.  They are not easy to get yet.  Also, CircuitPython has not been fully ported to the Pico W yet.

BjornTheProgrammer, 2022-09-18:
I have a couple of Pico W's. I would be more than willing to test and help out.

javierantunez, 2022-09-27:
Hey @dbisu maybe you can open sponsorship for this project, so interested parties can help to afford the expenses of development (ie the cost of the Pico W boards, your time, etc.) In hope that helps.

BjornTheProgrammer, 2022-09-29:
https://github.com/adafruit/circuitpython/pull/6933

It looks like WiFi support has been merged into main for circuit python.

dbisu, 2022-10-02:
It looks like CircuitPython 8.x will support the Pico W.  I should have a board soon. I expect it will take a couple of weeks for me to add support for some wireless features.

javierantunez, 2022-10-24:
Hi! everyone is there any new? @dbisu can we do something that helps you?

dbisu, 2022-10-25:
Sorry, I have not had much time to work on this project recently.  Hopefully over the next few weeks I'll get a chance to update to CircuitPython 8.0 (beta right now) and get the code ported.
Sounds like AP mode doesn't work yet, so I'll have to change a few things to handle that.

javierantunez, 2022-10-25:
Dave Thanks for the update!

Careless115, 2022-11-17:
I am trying to use the Pico W as an access Point which loads a webpage. I have added buttons and functions to my Micropython script. Now i am trying to run a script on the host pc if that button is pressed. I don't think anyone has done it yet but I do think that it is possible. Can you help?

dbisu, 2022-11-20:
I've create a branch called picow for support of the Pico W board.  The start_ap function doesn't work yet.  The website hasn't been tested, so consider all of this code as experimental for now.
The code needs CircuitPython 8.0.0-beta4 or newer.

DienoX, 2022-12-22:
Has anything changed since then?

vithp, 2022-12-29:
> Has anything changed since then?

yep exactly 3 weeks ago they added start_ap() from jepler into circuitpython beta 5
and a week ago beta 6 came out so today i decided to work on it.

i have gotten the regular ducky to work on the pi cow with modification and the idea is to edit the payload.dd file from a webserver running off of the pi cow

dbisu, 2022-12-29:
I have about half the code ported over to the Pico W.  Now that beta 6 has the wifi working, I should be able to get a test version going this week.

vithp, 2022-12-29:
Great. Thats good news for me since i only own a Pico W. Now i guess i can cause chaos with a usb.

dbisu, 2022-12-29:
I have updated the picow branch.
Changes:
working website on Pico W

I still have some other changes planned for differences between pico and pico w.

I also need to make the webpages look better.

I don't intend to merge this branch until CircuitPython 8.0.0 is officially released.

linuzzx, 2023-01-22:
> I also need to make the webpages look better.

I know css if you need help with that

closed 2023-02-18 by dbisu