dbisu/pico-ducky #82: When I drag the uf2 file into the pico i see only a "boot_out" file

p-i-c-o, 2022-07-08:
**Describe the bug**
When I drag the uf2 file into the pico i see only a "boot_out" file

**To Reproduce**
Steps to reproduce the behavior:
1. BOOTSEL button and connect
2. drag uf2 in root
3. see only "boot_out" file

**Expected behavior**
From a screenshot of a video from NetworkChuck, I can see he has these files, which I don't have.

NetworkChuck:
[image]

Me:
[image]

**Additional context**
I have used this specific pico before for this specific project, but have meddled too much with the files.

dbisu, 2022-07-08:
I've seen this before.  The way to fix is to download the reset firmware using the [flash_nuke.uf2](https://datasheets.raspberrypi.com/soft/flash_nuke.uf2)
Then, hold BOOTSEL to get to the RPI-RP2 USB mount point and follow the normal setup instructions.

official-hack4peace, 2022-08-06:
Can you explain in detail.

dbisu, 2022-08-06:
Steps to recover a Raspberry Pi Pico:
1) Download [flash_nuke.uf2](https://datasheets.raspberrypi.com/soft/flash_nuke.uf2)
2) While holding the BOOTSEL button on the Pico, plug in the USB cable to your computer.
3) When the RPI-RP2 drive shows up on your computer, copy the `flash_nuke.uf2` file to the Pico
4) After the device reboots, follow the Install instructions [here](https://github.com/dbisu/pico-ducky/blob/main/README.md)

closed 2022-09-18 by dbisu