dbisu/pico-ducky #84: disable usb drive

ratarchivos, 2022-07-24:
how i can put a counter that enable usb drive at 20 secons i made i and i put in code.py but did't work

[image]

dbisu, 2022-07-24:
Unfortunately, the way the storage module works in CircuitPython, you can only change the USB state in boot.py and not in code.py.

ratarchivos, 2022-07-25:
> Unfortunately, the way the storage module works in CircuitPython, you can only change the USB state in boot.py and not in code.py.

with a import boot.py or from boot.py import usbDrive
meaby i can put that counter and when finish change the boot.py to false

dbisu, 2022-07-25:
boot.py is only executed on cold boot, so changing this file will only take effect the next time the device is power cycled.

dbisu, 2022-09-18:
Closing this issue, not possible to change USB status in CircuitPython in code.py, only boot.py.

closed 2022-09-18 by dbisu