dbisu/pico-ducky #89: Feature: Hak5's Keystroke Reflection Exfoliation

bluegizmo83, 2022-08-10:
Not a big, just a feature request.

It would be awesome if you could include Hak5's new Rubber Ducky Keystroke Reflection Exfiltration technique! It allows data Exfiltration WITHOUT the need for Mass Storage or Network. It works by converting your exfil data (text only? not sure) to a binary data stream and then toggling the keyboard LEDs (Caps Lock and Scroll Lock I think) to represent the 1's and 0's of the data stream, and the Rubber Ducky see's any keyboard status LED changes so it decodes and saves the the binary stream to the Rubber Ducky right along side the original payload (actually, it might store it as raw binary and you manual decode if afterwards, I'm not sure), without the Ducky ever having to be mounted as mass storage during the attack!

bluegizmo83, 2022-08-12:
> Looks like the peripheral-side could be implemented by checking `kbd.led_on(x)`, possibly sending two bits and a clock at once

Yeah I believe that's how it's implemented in Hak5's Ducky as well. I read something about it watches the LED for a specific start sequence and then records the data.

dbisu, 2022-08-26:
I've been looking at the HID API.  I'm not sure yet if there is a way in the current API to get the data back from the host system.
Doing the exfil via the keyboard LEDs looks fairly straight forward.  I haven't looked at the Hak5 info on the coding they use to send and read the data as an attacker yet.

beamzer, 2022-12-10:
Just read the whitepaper on: [HAK5](https://cdn.shopify.com/s/files/1/0068/2142/files/hak5-whitepaper-keystroke-reflection.pdf) and they (probably) are using the numlock and capslock states, since those are sent from the computer to the keyboard. So numlock for the 1's and capslock for 0's or vice versa. Simple but effective idea.

che3sy, 2025-09-02:
I know it's been a long time, but I created a fork and added those features. It does have some caveats, though.
Because of the issue with the while file writing (you can't write files to the Pico storage while it's in HID mode and visible to the host), when you first plug in the Pico with the $_EXFIL_MODE_ENABLED set to true, it will not show up as a drive. Once the loot is saved, (uses scroll lock to signify end of data) the loot.bin file will be created, reinsert the Pico into the computer and it will now show up as a drive, allowing the user to examine the loot, the payload will also not run if a loot.bin exists.