dbisu/pico-ducky #94: Assets on Root directory

LyQuid12, 2022-09-08:
Is it possible to put assets on Pico's root directory? So when run on a computer, it doesn't need to download the assets, just copy from root directory.

LyQuid12, 2022-09-09:
Nvm, it works

closed 2022-09-09 by LyQuid12

d3k4-k3rb3r0s, 2022-09-09:
err, elaborate? that sounds like a game changer..

dbisu, 2022-09-09:
It is possible to store some files locally on the USB mass storage part of the Pico.  There is, I believe, less than 1MB of storage available, so your assets need to be fairly small.
You need to have a good idea on where it will mount/enumerate on your target computer for the scripts to work properly.

d3k4-k3rb3r0s, 2022-09-09:
interesting. the least appealing thing about ducks is that we almost always have to rely on having the target reach out and pull down the true payload.

LyQuid12, 2022-09-09:
> It is possible to store some files locally on the USB mass storage part of the Pico. There is, I believe, less than 1MB of storage available, so your assets need to be fairly small. You need to have [...]

Yea, i have to compress my bmp file which was originally 9 mb to 200kb :|