dbisu/pico-ducky #100: USB enable/disable mode

MrIslamAIman, 2022-09-20:
I have a little problem with the USB enable/disable mode 
when I connect pin 18 (GND) and pin 20 (GPIO15) it does nothing the USB work like storage and the script works perfectly 
I thought the problem was with the jumper....  but when I connected pin 1 (GP0) to pin 3 (GND) I entered the Setup mode with no problem 

I searched in the other problems I found no one has this problem...... I hope u help me guys 
BTW it happens to my 3 raspberrys

p-i-c-o, 2022-09-26:
Check if you have boot.py on root, and re-install everything. An issue that I pointed out just got fixed. Good luck

dbisu, 2022-09-26:
Yes, that is likely the problem.  I will update the main instructions to include the copying of boot.py by default.

dbisu, 2022-09-27:
I've updated the instructions now to include the copying of boot.py
Hope that helps.

closed 2022-11-01 by dbisu