dbisu/pico-ducky #101: Not quite sure what I did wrong.

SoldierBoy123, 2022-09-21:
After all the steps, plugged it in, not working. does nothing
Pops up as a removable device.... and then..... nothing. I wanted to rick roll my sister, but nothing happened.
Now I am new to this, I'm just starting to learn my own way with this stuff, so if someone can tell me if I have to restart the whole process, or erase it all? let me know please.

Endercreepe, 2022-09-23:
Welp I have the same problem. It worked for years and suddenly it just stopped working. I

dbisu, 2022-09-23:
Best bet is to follow the instructions in RESET.md
https://github.com/dbisu/pico-ducky/blob/main/RESET.md

Endercreepe, 2022-09-24:
Well, I did exactly that and it didn't work.
At least I think I know the Issue. If I open code.py in an IDE it says it can't reach the 
Device although its clearly plugged in.
So @dbisu could we maybe do a discord call where I show you the Problem? If yes could you send me your tag?

dbisu, 2022-09-24:
Sorry, I've got some other things I'm working on this weekend.
A option would be to include these items from the issue template.  Also post pictures of your board.
If possible, include debug serial data. On Windows, use PuTTY, connect to the debug serial port (commonly COM3, but could vary)

On Linux, use minicom (minicom -b 115200 -o -D /dev/ttyACM0, where ttyACM0 corresponds to your device)

Thanks.

SoldierBoy123, 2022-09-24:
Hey btw, idk if this is what your working on, but there is a back door
exfiltration script that would not work for me. Everytime it tried to get
administrative access it would click yes instead of no

Endercreepe, 2022-09-25:
[image]
Here

dbisu, 2022-09-25:
Do you have debug serial logs from the board?

Endercreepe, 2022-09-29:
Sry for being that late, but und imma aend that tomorrow.

SoldierBoy123, 2022-10-11:
I think I know what happened with me btw. It was the files, I had to go
back and thoroughly make sure I copied the right files

dbisu, 2022-10-12:
Closing, issue resolved.

closed 2022-10-12 by dbisu