dbisu/pico-ducky #102: 22 september 2022 - i can verify that this repo works on windows 10

qkum, 2022-09-22:
most pentest repo's for raspberry pi dont work or only partly works 2022.

this project is blessed with both up-to-date working code + documentation that ranks 5 out of 5 stars.
great work, you deserve a medal for your work @dbisu 🥇 
how is the new generation of white-hats going to learn anything with the trashy documentation most projects provide?

they could learn a lot from you fren.

**btw:**
"Download [CircuitPython for the Raspberry Pi Pico](https://circuitpython.org/board/raspberry_pi_pico/). *Updated to 7.0.0"
i used version 7.3.3 and it worked fine.

qkum, 2022-09-22:
when new pentesters struggle big time or have to give up learning this craft...because the documentation suck...you are doing the criminals a favor and try guess who is going to secure the world when you are on pension? not the thousands of people's time you wasted with half-assed documentation or code.

qkum, 2022-09-22:
i have personally wasted countless hours on code that did not work or had shitty documentation

and my learning has been delayed by years as a result

but this project is one of the rare exceptions to the rule when we talk raspberry pi / pico projects

qkum, 2022-09-22:
afters years of hard work i can finally control/fix my headless raspberry pi's by plugging in a badusb

now i can save 100+ hours of problem shooting for the rest of my life using my secure & homemade badusb

not everybody got 200+ usd to throw after a finished solution to have good old china as counterparty risk...

with raspberry pi products you know waht you get.

dbisu, 2022-11-01:
Thanks for the comments.
Closing.

closed 2022-11-01 by dbisu