dbisu/pico-ducky #106: raspberry pico w

kazikaas, 2022-10-01:
do you already have a script for the raspberry pico w?

p-i-c-o, 2022-10-01:
I have a feeling that the current code should work with the new chip

dbisu, 2022-10-02:
I'm still working on getting a pico w. I should have one soon.  I expect then it will take a couple of weeks to make the updates to support the wireless features.
I appears that CircuitPython 8.x will support the Pico W.

lorelolorelo, 2022-10-20:
I was playing with the code that you have available here,`https://github.com/dbisu/feathers2-ducky` it seems that adapting it is not very complicated even though I am not an expert.

In any case, it loads all the code except for the webpage, I've never deal with wsgi before, but it seems that some regex or similar is not right.

The only things I have changed is `wifi.radio.ipv4_address_ap` for `wifi.radio.ipv4_address`

And it seems AP mode is not available yet for circuit python so this line won't work... Sugestions are welcome in any case i plan to spend some more time on the wekeend :)

dbisu, 2022-11-01:
Closing, duplicate of #80 
I have a Pico W now, just need some time to get it working.

closed 2022-11-01 by dbisu