dbisu/pico-ducky #107: not working

rbctv1, 2022-10-03:
hi i wanted to try the bad usb myself so i bought the rasbery pi pico and did all the things from the totorial but when i put in the payload nothing is happening?

p-i-c-o, 2022-10-03:
Hi there!

Here are some steps I found can be confusing:

1: The 'asyncio'  folder has to have at least 7 files in it, there is an empty folder named the same thing that only has a 'requirements.txt' file in it.

2: Make sure to have these folders or files in your Picos lib folder:
- adafruit_hid
- adafruit_debouncer.mpy
- adafruit_ticks.mpy
- asyncio

And finally make sure your script is properly written and has no syntax errors:

Here's my usual test script.

```
REM Test Payload
DELAY 500
STRING Hello World,
DELAY 500
ENTER
DELAY 500
STRING Good to see you!
DELAY 500
REM End
```

dbisu, 2022-10-03:
Probably the easiest way to get started is to use one of the release bundles and follow the instructions on the Releases page
https://github.com/dbisu/pico-ducky/releases/tag/v1.4
If that doesn't work, follow the outline in the issues template to gather more debug information.

bughunter151, 2022-10-08:
> hi i wanted to try the bad usb myself so i bought the rasbery pi pico and did all the things from the totorial but when i put in the payload nothing is happening?

I tried it now follow the steps entioned in readme & it will work for you

rbctv1, 2022-10-08:
Yea it works now thanks

dbisu, 2022-10-09:
Closing, issue resolved.

closed 2022-10-09 by dbisu