dbisu/pico-ducky #108: Was working on wednesday not working now

iescobar76, 2022-10-14:
Hello guys hope someone can enlighten me.  I have used this SW since it originally came out 2 years ago.  I used it on wednesday to show to my high school students how to make their own USB rubber ducky's. It worked fine on wednesday.  Today i showed them on a follow the teacher step by step and i used both payload studio and ducktoolkit to encode a simple hello world on my mac, just like two days ago.  Well the modified raspberry pi pico does not run the payload.dd (got the inject.bin renamed as usual) and nothing.  Checked again, checked again.  Nothing.  Used the payload.dd i created same way last wednesday and it works.  Used payloads i created 1 year ago and it works.  Use the generated inject.bin on a USB rubber ducky and it works.  

Any ideas?
cheers
ivan

dbisu, 2022-10-22:
If possible, please include information from the debug serial port.
Another option could be go through the instructions in RESET.md.

dbisu, 2022-11-01:
@iescobar76 Any more information on this issue?

iescobar76, 2022-11-01:
Got it fixed sorry for the bother I had a virus creeping out my pc that’s why it was failing
Cheers ivan

dbisu, 2022-11-01:
No bother, just trying to clean up old issues.

closed 2022-11-01 by dbisu