dbisu/pico-ducky #109: Pico Not Working

lucasB69, 2022-10-29:
i have followed the instructions but its not working for some reason please help!

[image]
[image]
[image]

WarpOverload, 2022-10-29:
Same thing on my end, except I'm running Arch Linux. I tried bombing the flash and following the steps again, it didnt work. Maybe try using an older release?

WarpOverload, 2022-10-29:
I tried using pico-ducky 1.3 and everything went smoothly. Try that and if it works, I guess you should just wait till the maintainers of this repository fix the issue.

dbisu, 2022-10-30:
@lucasB69 Can you be more specific?  What isn't working?  If you bring up a serial terminal, are you getting any errors?

lucasB69, 2022-10-30:
@dbisu thanks for responding but i fixed it!

dbisu, 2022-10-30:
OK, great. Would you mind putting in a comment on how you fixed it as a closing comment?
Thanks.

zigouris, 2022-10-31:
> @dbisu thanks for responding but i fixed it!

How did you fixed it cause i have exactly the same problem?

dbisu, 2022-11-11:
> same

Can you be more specific as to what is not working?

dbisu, 2022-12-09:
Closing.  The original issue has been resolved.

closed 2022-12-09 by dbisu