dbisu/pico-ducky #110: Please someone help me it does not work

zigouris, 2022-10-31:
I have followed all the steps and it just does not run
I mean that i have put all the necessary files in the lib folder
[image]

I have all the necessary files in the root too
[image]

**Expected behavior**
I expected to run the payload.dd file when i injected the pi pico in my pc but it does not

please someone help me i would appreciate it (i have all the correct versions!)

zigouris, 2022-10-31:
z1g#0069 here's my discord too if you could send me a message to fix this problem it would be better but we can fix that here too!

dbisu, 2022-10-31:
Are you using a US keyboard or a non-US keyboard layout?

zigouris, 2022-11-01:
@dbisu  An us keyboard layout

dbisu, 2022-11-03:
I attempted to reproduce tonight and was not able to recreate any issue.
I blanked the Pico using flash_nuke.uf2
I then copied all of the files from the US release v1.4
The pico ducky behaved as expected.

If possible, using PuTTY, copy the information from the debug serial console.

vithp, 2022-11-05:
is it a pico w? pico w does not have the latest stable version therefore it cant run pico-ducky

dbisu, 2022-12-09:
Have you tried using one of the release bundles?

dbisu, 2023-02-18:
Closing, no activity.

closed 2023-02-18 by dbisu