dbisu/pico-ducky #111: Not an issue : possible hardware improvement ; RF picoducky

one-twoo, 2022-11-03:
so i have salvaged these from an old wireless keyboard and mouse , the keyboard have an RF emitter @ 2.4 Ghz , and the reciver part is a little dongle working @ the same frequency and channel :
[image]
my idea is to connect the pico with the emitter part , and just plug the little dongle in the target system instead of plugging the bulcky pico 
[image]

advantages : more stealthy , and we can launch the payload from distance ( with the push button) 
the pico would be battery powered , the transmitter part connected with the usb port of the pico or GP pins ..., once the dongle is inserted in the target the pico should be immidiatly paired since the transmitter and reciever operates @ the same specific channel and frequency 
i dont know where to start or even if this is possible so im asking for a hint!

andreasbrett, 2022-11-04:
don't think that would work but definitely interesting idea to make a more stealthy ducky

one-twoo, 2022-11-06:
> don't think that would work but definitely interesting idea to make a more stealthy ducky

hey andrea , ive seen alot of people transforming regular wired devices into wireless devices via bluetooth  modules , universal usb to wireless adapters and other RF emitters-reciever combo , im still convinced this is possible , at least the idea is good , unfortunately i dont have the necessary skills to do it , i hope someone think about it

dbisu, 2022-11-07:
I think this is an interesting idea.  However, I see the idea as out of scope for the pico-ducky project.
In terms of wireless, support for the Pico W will be coming (hopefully) soon.

one-twoo, 2022-11-07:
> I think this is an interesting idea. However, I see the idea as out of scope for the pico-ducky project. In terms of wireless, support for the Pico W will be coming (hopefully) soon.

thanks dbisu ! i hope you will be done soon with the new W

one-twoo, 2022-11-08:
i didnt want to open a new issue for this , i ditched the old hardware and got the pico w , to my surprise after i flashed it with the latest release of circuitpython 8.0.0.beta4 i found that it had very little storage space : barely 490 kb 
is this normal , i had almost 1 mb with the regular pico , with that space i used to host 4 payloads plus all the assets needed to perform attacks without internet connection and exfiltrate data into the board , this could be a problem now ..
[image]

dbisu, 2022-11-11:
You'd have to ask in the adafruit discord about the size of the circuitpython releases.
The pico W has more functions with the wireless, so probably needs a larger image.

dbisu, 2022-12-09:
Closing.  Pico W support is covered in a different issue/feature request.

closed 2022-12-09 by dbisu