dbisu/pico-ducky #112: every thing working fine but payload is not executing

bharathdpv, 2022-11-06:
[image]

see the payload is not working in my pc

one-twoo, 2022-11-06:
what's your payload.dd look like ?
what's your keyboard layout and did you added the necessary keyboard libraries ?
have you added a push button or payload selector to your pico ? and if you modified your board whats the situation of the modified pins?
do you have boot.py in the root of the pico?

bharathdpv, 2022-11-11:
ohhh so much question i will answer!!

my payload.dd look like:
[image]

my keyboard layout:
[image]

yes i added all the nessory library 
[image]

yes i have boot.py in my root !!

payload selector is in the code !!   no i did not modified any pins

dbisu, 2022-11-11:
I tested this payload on a pico board here and it worked for me(tm).
Try adding this line after line 168:
`print(line)`
and then capture the debug serial output.

bharathdpv, 2022-11-12:
[image]

NOTHING WORKS!!!

dbisu, 2022-11-12:
Have you tried blanking the pico and installing from the release v1.4 zip?
Also, did you modify other lines?  The code should go into a loop and not go to the `Code done running` statement without stopping the program.

bharathdpv, 2022-11-12:
Yes I modified the lines  because only the led up and down no other lines showing in the debug console so, I modified for my use and the code works fine but after few  time working it stoped working(I also changed the pc and Android phone even cable).

dbisu, 2022-11-12:
I am unable to reproduce this issue.
My recommendation would be to start over with the v1.4 image.

bharathdpv, 2022-11-12:
reinstalled  v1.4 same issue

https://user-images.githubusercontent.com/101401135/201456252-674d3357-0983-4199-b315-82268d786498.mp4

check out this video!!

dbisu, 2022-11-12:
If you unplug the device from your computer and plug it back in, does it run the payload?  This would be without using the IDE.

bharathdpv, 2022-11-12:
nah nothing works only the led blinking and the circuitpy window popping off!!

dbisu, 2022-11-12:
I am unfortunately unable to reproduce on the boards I have here.
From what you have shared, there are no errors running the code and it is able to find your payload.
If you run the step manually, do they work properly on you computer?

bharathdpv, 2022-11-12:
It works some time and it won't work, last week it works fine but now it's not working

dbisu, 2022-11-12:
I've not seen that issue.  Based on the information provided, I would expect it to be working.

dbisu, 2023-02-18:
Closing, not active.

closed 2023-02-18 by dbisu