dbisu/pico-ducky #114: wait_for_button_press Command Not Working

PixellNerd, 2022-11-15:
When the script runs, it skips over any `wait_for_button_press` commands. I have tested my own scrips as well as the example scrips from the Rubber Ducky documentation (images below respectively)
[image]
[image]

As far as I can tell, there shouldn't be any button pressed down that would cause it to be skipped over.
Any help would be appreciated!

dbisu, 2022-11-16:
WAIT_FOR_BUTTON_PRESS is not currently implemented.  See issue #103

dbisu, 2023-02-18:
Closing as duplicate of #103

closed 2023-02-18 by dbisu