dbisu/pico-ducky #117: No functions, loops and variables?

iamSkev, 2022-11-26:
I've just got a pico and wanted to try writing my own duckyscripts and wanted to say Hello World 10 times in notepad. It never wrote anything when it got to notepad

```
GUI r
DELAY 800
STRING notepad
DELAY 600
ENTER
VAR $FOO = 10
WHILE ( $FOO > 0 )
    DELAY 700
    STRING Hello World
    DELAY 800
    ENTER
    $FOO = ( $FOO - 1 )
END_WHILE
```

dbisu, 2022-11-26:
Loops are part of Ducky Script 3.0, which isn't currently supported by this project yet.  It is on the to-do list, though no timeline yet on when I'll get it done.

iamSkev, 2022-11-26:
> Loops are part of Ducky Script 3.0, which isn't currently supported by this project yet. It is on the to-do list, though no timeline yet on when I'll get it done.

Ah i see, i just wanted to check if it was supported and didn't just miss something while looking for it in the source code. I guess i'll just copy it multiple times to get the same effect.

dbisu, 2022-12-12:
I'm not sure yet. Hak5 recently released a whitepaper on how some of those functions work.  I have not evaluated yet how hard they will be to implement in this codebase.

SimonB-cyber, 2022-12-23:
Wo kann ich den Raspberr pi pico projekte finden? Also fertig gecodete Datein?

KsAmJ, 2023-03-11:
a quick walkaround if you want a script to run forever then you can modify the code on the code.py 

```
if(progStatus == False):
    # not in setup mode, inject the payload
    payload = selectPayload()
    print("Running ", payload)
    while True:
        runScript(payload)
        time.sleep(xxxx)
    print("Done")
else:
    print("Update your payload")
```

change xxxx to total seconds time which exceeds the total delays in the payload.dd

dbisu, 2024-09-21:
Support for while loops was just merged into main.

closed 2024-09-21 by dbisu