dbisu/pico-ducky #118: Added all files, but no payload execution.

MadeinHeavan, 2022-12-07:
Ive got the correct keyboard, and all files needed. I am running 7.0.0 adafruit and got the bundle 7. 

Any help would be appreciated.

MadeinHeavan, 2022-12-07:
[image]
[image]

dbisu, 2022-12-08:
The first thing I see is that adafruit_tick should be a mpy file and not a folder.
You may want to download one of the release bundles to ensure all of your library dependencies are correct.

MadeinHeavan, 2022-12-08:
thank you, I will try.

MadeinHeavan, 2022-12-08:
it worked, thank you so much. btw I used a pre built release.

dbisu, 2022-12-09:
Closing since it is working now.

closed 2022-12-09 by dbisu