dbisu/pico-ducky #121: Changing the default pins for payloads

alext30Developer, 2022-12-16:
Hi! I noticed that in the 'code.py' file it's mentioned that you can switch between 4 payloads by connecting GPIO pins 4, 5, 10, 11 to ground. What code would I need to modify to change those pins like this:

Payload 1: GP16 to GND
Payload 2: GP4 to GND
Payload 3: GP11 to GND
Payload 4: GP20 to GND
I attached a picture of what it would look like.
[image]
Also, I'd love to see details for multiple payloads on the readme, as I accidentally stumbled across it by reading the script. Thanks in advance!

dbisu, 2023-02-18:
In the code that is about to be merged, duckyinpython.py lines 115 to 123 have the GPIO settings.  The current pins match a PCB I made for this project.  Feel free to use whatever pins work for your projects.

dbisu, 2023-02-18:
I've also updated the README to include some info on the multiple payloads.
Thanks.

closed 2023-02-18 by dbisu