dbisu/pico-ducky #123: Hi! Please help a noob.

0067sdsm, 2022-12-25:
So i tried to make it work lol, and i think everything is as it should but for some reason it doesnt work? I plug the pico in and it only shoes up as a mass storage device
(screen shot for reference:)
[image]

alext30Developer, 2022-12-28:
Hi! I can see that you are missing the boot.py script, copy it over from the repository (step 9). Also verify that in the lib folder you got the following things: 
[image]
If you do, that's great! But if you don't, check out steps 5, 6, 7, 8.
The rest looks good. Let me know if that fixed it for you.

0067sdsm, 2022-12-28:
Thanks! Will do when I get home.

alext30Developer, 2022-12-29:
Hi again! Did it work this time?

0067sdsm, 2022-12-29:
I'm in a completely different country XD I'm sorry, ill let you know tho

0067sdsm, 2022-12-29:
I'm pretty sure that in the lib I only had the adafruit_hid so that's probably it (+ the boot thingy) could you let me know where that stuff is just so when I get back I'd o t have to go looking through all the files and tutorials lol

dbisu, 2022-12-29:
Look at the zip files here:
https://github.com/dbisu/pico-ducky/releases/tag/v1.4
Follow the instructions on this page and you should have a ready to go pico board.

0067sdsm, 2023-01-02:
WORKS!!! this is so efing awsome thankyou!!!

alext30Developer, 2023-01-02:
I'm glad it worked! Remember to close the issue btw

closed 2023-01-02 by 0067sdsm