dbisu/pico-ducky #124: Payload is not getting executed after changing the language to german

alex020302, 2022-12-25:
Hello,
when I still had the US keyboard enabled, everything worked and the payload was executed successfully. Now I need special characters, which is why I absolutely need the German keycodes. Again, I have set everything as in the instructions, changed and adopted. Nevertheless it does not work. Can someone please help me? I guess I have to change the "duckyCommands". Here are screenshots:

[image]
[image]
[image]

dbisu, 2022-12-31:
Have you tried using the German release bundle?
https://github.com/dbisu/pico-ducky/releases/download/v1.4/pico-ducky-v1.4_win_de.zip

dbisu, 2023-02-18:
Closing, no activity.

closed 2023-02-18 by dbisu