dbisu/pico-ducky #126: Twin-Duck Firmware

gdz11223344, 2023-01-04:
I have a question, a payload I'm going to use need twin-duck firmware. So does pico-ducky allow it?

dbisu, 2023-01-04:
Kind of.  If you connect a button to GPIO22 and connect a jumper from GPIO0 to GND, you will get close.
the payload will not deploy when plugged in, and then will run when you press the button.

gdz11223344, 2023-01-04:
So, the payload will start to work when the button is pressed? And there isnt a more discrete way of doing it? (btw this is for a joke not something serious)

gdz11223344, 2023-01-04:
So, the payload will start when the button is pressed? and also could you show a diagram? I am new with this thing

dbisu, 2023-01-05:
yes, the payload will start when the button is pressed.  I don't have a mock-up of a button.

dbisu, 2023-02-23:
Closing, button capability added.  Support for the WAIT_FOR_BUTTON command is covered in a different issue.

closed 2023-02-23 by dbisu