dbisu/pico-ducky #127: Other modules

1RaY-1, 2023-01-04:
Hello, i have 2 questions.
1. Can I use additional modules in duckyinpython.py (code.py) that have to be installed with ```pip```, for example ```keyboard```, to be able to detect key presses, or any other module?
As I understand additional modules have to be in ```.mpy``` extension, but I never used this extension, so don't understand.

2. When I plug in the pico, how can I make it just run python code (code.py) without any payloads, to directly perform python scripting.
I tried to write a simple python script to create a folder on my desktop (using os.mkdir) but it didn't work.

Thanks

dbisu, 2023-01-04:
The biggest thing to keep in mind is that the code is running on the Pi Pico and not your host computer.  That is why your calls to os.mkdir didn't do anything on your computer.
This project configures the Pi Pico to act as a USB keyboard (HID) and inject key strokes to the target computer.

1RaY-1, 2023-01-05:
The thing is that in Circuit Python download page in *Built-in modules available:* it says ```os``` that's why i was using specially this module.

dbisu, 2023-01-05:
no, it will not. Again the python code is running on the Pi Pico CPU, not your host/target computer.
Pico-ducky works by acting like a keyboard.  The payload files are basically instructions on key strokes to send to your computer.
CircuitPython is a fairly robust environment, so it includes modules like os to be able to run some interesting programs on the Pico.

1RaY-1, 2023-01-05:
Ok.
Thanks

closed 2023-01-05 by 1RaY-1