dbisu/pico-ducky #132: CircuitPython

YashasSingh, 2023-02-02:
Does CircuitPython 7.3.3 work with everything else or do I have to do something extra for it to work? I am asking because after following all steps it is not working even with my own code and the instructions says it only goes  to 7.0.0  so will this plan still work.

dbisu, 2023-02-05:
Yes, this works on 7.3.3.  The release bundles include that version of CircuitPython.

skyblaster, 2023-02-07:
I just picked up a YD-RP2040 which is only supported by 8.0.0.
I followed the install instructions but using the adafruit-circuitpython-bundle-**8.x**-mpy-20230206.zip bundle and unfortunately couldn't make it work. Is there any experimental code to try?

skyblaster, 2023-02-16:
Found Wokwi perusing the forks of this project. Very cool.
While my example is quite basic, I've partially replicated the [YD-RP2040](https://circuitpython.org/board/vcc_gnd_yd_rp2040/) hardware on alternate GPIOs
https://wokwi.com/projects/356786290501324801

Cycling through payloads with a rotary encoder and then executing it with the built in USR button is my next goal. Anyone with better coding skills want to share that code :-)

dbisu, 2023-02-18:
There is a PR in process to update support for CircuitPython 8.0.0

skyblaster, 2023-02-19:
Here's my rudimentary rotary encoder example if anyone wants to scrutinize it while I wait for mine in the mail.
https://wokwi.com/projects/357021888172298241

dbisu, 2023-02-23:
Closing, original question is answered.

closed 2023-02-23 by dbisu