dbisu/pico-ducky #133: it does not work nothing happens does the computer need python

h54t45tv45, 2023-02-03:
**Describe the issue**
A clear and concise description of the symptoms

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

YashasSingh, 2023-02-03:
I honestly don't think it needs python script rather the adafruit-circuitpython-bundle-7.x-mpy-YYYYMMDD.zip is not the right one  to download  
use https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases?page=7
as this is the one that works

quick question do you guys know why this pico nuke is not working 

https://user-images.githubusercontent.com/121969113/216629868-f204881c-9eb8-46da-af64-de127f1c7b54.mp4

dbisu, 2023-02-05:
The host or target computer does not need python. CircuitPython runs on the Pi Pico.

beammedown, 2023-02-11:
@YashasSingh For me, the Nuke works after plugging the pico in while pressing the "BOOTSEL" Button.

So you probably have to press "BOOTSEL" and plug in the pico. After the pico appears, you should just be able to copy the nuke over and have it working. It should quickly disconnect and then reconnect.

dbisu, 2023-02-18:
Were you trying to use CircuitPython 8.0.0?  If so, I have PR in process that updates support for 8.0.0

dbisu, 2023-02-19:
If you are trying to use the flash_nuke image, please follow the instructions in RESET.md.
As beammedown noted, you need to be in the bootloader mode by holding the BOOTSEL button and having the Pico show up as RPI-RP2.

dbisu, 2023-02-23:
Closing, support for CircuitPython 8.0.0 has been integrated into main.

closed 2023-02-23 by dbisu