dbisu/pico-ducky #136: CIRCUITPY folder opens but payload does not execute

lilkahuna, 2023-02-14:
I have recently followed the tutorial using 2 different zips of circuit python(adafruit-circuitpython-bundle-7.x-mpy-20221215 and adafruit-circuitpython-bundle-7.x-mpy-20230213). I have used a network chuck tutorial and that didnt work so i used the nuke.uf2 to restart and try again. I ended up following the instructions on the github step by step try the diffent zips and libraries, i have had no luck. The board I am using is the RP2040.

Does anyone have an idea on what to do?
Thanks you.

skyblaster, 2023-02-16:
If you haven't done so already, I would recommend installing the Mu editor so you can evaluate your code execution via the serial console:
https://learn.adafruit.com/welcome-to-circuitpython/installing-mu-editor

elliotmoreau, 2023-02-16:
I have the same exact issue, I also first used network chuck tutorial and then ended up following the instructions on the github but it's still not working...

If anyone could help that would be cool.

elliotmoreau, 2023-02-16:
Ok I found the problem for me, this line : # notsupervisor.disable_autoreload() seems to make the script crash I removed it and now it works.

lilkahuna, 2023-02-16:
Does it actually run th script when plugged into the pc? Also what operating system do you use because I was rold it doesn't work on windows anymore because of a recent patch.

elliotmoreau, 2023-02-16:
> Does it actually run th script when plugged into the pc? Also what operating system do you use because I was rold it doesn't work on windows anymore because of a recent patch.

I'm on windows 10 and yes it runs when I plug it in !

lilkahuna, 2023-02-16:
Alright, did you just use the newest version of circuit python and follow the instructions? Just want to make sure I'm doing it right this time.

lilkahuna, 2023-02-17:
What version of circuit python do you use and should I just put code and boot.py in the  root of the Pico right?

dbisu, 2023-02-17:
Hey all,
I was out because of a work trip. I am still working on validating the code with CircuitPython 8.0.0.  I'm hoping by the end of the weekend to have any changes merged into main.

elliotmoreau, 2023-02-17:
> What version of circuit python do you use and should I just put code and boot.py in the root of the Pico right?

The 7x one and yes code.py and boot.py should be in the root of the pico

elliotmoreau, 2023-02-17:
> Hey all, I was out because of a work trip. I am still working on validating the code with CircuitPython 8.0.0. I'm hoping by the end of the weekend to have any changes merged into main.

I don't really know what it's gonna change to be honnest but I mean if it fixes the issue it's cool ! I just bough a pico 2 days ago to try this so don't really know about CircuitPython and all this stuff ahah.

lilkahuna, 2023-02-17:
I used your lib files and 7.0.0 circuitpy but it still won't work. I also commented out line 127.

elliotmoreau, 2023-02-18:
> I used your lib files and 7.0.0 circuitpy but it still won't work. I also commented out line 127.

Execute your code.py in this : [https://codewith.mu/](url) If there is any error just tell me. That's how I found the problem with the line 127.

dbisu, 2023-02-18:
If you change line 127 to this, it will give the same function, updated for CircuitPython 8.0.0
```
# turn off automatically reloading when files are written to the pico
#supervisor.disable_autoreload()
supervisor.runtime.autoreload = False
```
This is in the picow branch that I am currently testing for merge into main.

closed 2023-02-18 by dbisu