dbisu/pico-ducky #139: Other Keyboard Layouts don't seem to work anymore

linuzzx, 2023-02-19:
I made a script in the web GUI with "STRING y". I did install the correct lang files and modified the code.py as instructed

dbisu, 2023-02-19:
I think I may know the issue.  try commenting out lines 11 &12 in code.py and then making the correct changes for non-US keyboards in duckyinpython.py.

linuzzx, 2023-02-19:
I did

dbisu, 2023-02-19:
I think there is duplicated code.  all of the keyboard code should be in duckyinpython.py now.  keep everything commented out in code.py and make the changes in duckyinpython.py instead.

linuzzx, 2023-02-19:
Yep that worked, I removed lines 7-17 and did the stuff in duckyinpython.py. works thanks
you should probably remove that in the repo too

dbisu, 2023-02-19:
Yes, I'll make that change.  thanks for testing.

closed 2023-02-19 by dbisu