dbisu/pico-ducky #142: payload not running

GregboyGR, 2023-02-20:
hello, its been a week that i got in my hands the pi pico, i've spent more like 20 hours trying to download everything again and again to understand why the payload doesnt run. i've tried the version 7 and 8 too, the only difference that i find between these two is that in version 8 the light of my pi pico turns on and off while to the version 7 it does nothing, the tutorial that i tried is from networkchuck and after that i tried many times with your steps below your code. one thing that i tried is that generally its works with thonny that i also tried, but i dont know whats happenning this one. Plus i used a jumper wire just in case it wasn't in setup mode but nothing actually happens when i do it. Please i hope you could help me with my problem

dbisu, 2023-02-20:
Please include a dump of the debug serial to help determine the issue.

GregboyGR, 2023-02-21:
where can i find it, do you mean that i could find it from the python code? because if you mean that, i dont even think that python code runs either. anyway, i tried it i tried it manually with vs code and has some errors about usb_hid module.

dbisu, 2023-02-21:
The code will not resolve dependencies if loading locally in something like vs code.
If the LED is blinking, then the code is running on the Pico.

dbisu, 2023-02-22:
Use PuTTY or Mu editor to see what errors you are getting from the Pi Pico.
https://codewith.mu/en/tutorials/1.2/circuitpython

closed 2023-02-22 by GregboyGR