dbisu/pico-ducky #147: When I plug it in it only opens the file explorer.

Kyle2929, 2023-03-01:
When I plug in the pico, It only opens the file explorer and does nothing else. The code.py is not a python file and goes to notepad.

[image]

leggiand, 2023-03-01:
you have to copy boot.py,duckyinpython.py, code.py, webapp.py, wsgiserver.py from the project to the root folder of the pico

Kyle2929, 2023-03-01:
I was on an older release to see if it would work with it. I just put it back on the newest release and it still doesnt work.
[image]
[image]

dbisu, 2023-03-02:
Please use PuTTY or Mu editor to get the debug serial logs. This will help determine the issue.

dbisu, 2023-03-26:
Closing, no updates.

closed 2023-03-26 by dbisu