dbisu/pico-ducky #148: Cant connect to webservice

leggiand, 2023-03-01:
when i go to 
192.168.4.1:60 for connect to the webservice it says connection timeout

connect the pico when the payload is loaded and go to 192.168.4.1:80

to connect to the webservice

[image]

i also tried with and without payload on

dbisu, 2023-03-02:
verify that your computer connected to the Pico W access point using the ssid and password you put into secrets.py.

leggiand, 2023-03-02:
i check it and they're right

Th34rch1t3ctOfficial, 2023-03-05:
> i check it and they're right

And do you have device in the setup mode? Jumper between pin 1 (GP0) and pin 3 (GND). 

https://github.com/dbisu/pico-ducky#setup-mode

The pico-ducky doesn't run like the om.g plug, unfortunately. You must first run the device in the setup mode, then connect to the wifi AP you set in secrets.py, edit or create payload, unplug the jumper, and unplug it from the PC, then plug it to the PC again.

closed 2023-03-08 by leggiand