dbisu/pico-ducky #154: Wi-Fi section of bad usb

Ojamyjack, 2023-03-16:
I follow all the steps listed to set up my pico but when I attempt to connect to the web server I just don’t get any sort of response Is there any extra steps that may not be listed. I do have the pico w.

responsivereality, 2023-03-17:
Yep, I have the exact same problem. Would love to see a solution from the creator.

dbisu, 2023-03-17:
Please post a dump of the debug serial from the device.

The steps to use the wifi:
Make sure secrets.py has the AP name and password you want to create (Pico W acts like an AP).
Connect to the AP from your attacking device.
Go to http://192.168.4.1
You should see the main webpage for the pico-ducky webserver.

responsivereality, 2023-03-17:
Hey dbisu. Thanks for the prompt response. Unfortunately, my Raspberry Pi Pico W does not have any GPIO pins soldered, so I do not know how to get info from the debug serial. How do you setup "secrets.py"? There was nothing about that in the instruction page. Thanks for your helo.

dbisu, 2023-03-17:
The debug serial is available via the USB port. Use PuTTY, minicom, or Mu editor to access the debug port.

Ojamyjack, 2023-03-17:
> 

Where do I find secrets.py and set that file up

responsivereality, 2023-03-17:
Mu is not giving me any data. I cannot access the serial information, all it does is open a REPL.

dbisu, 2023-03-18:
> > 
> 
> Where do I find secrets.py and set that file up

I thought the README had that information.  I've updated README.md to include how to create secrets.py.

responsivereality, 2023-03-18:
Thank you so much! It now works as intended, except for one minor thing: When I upload a payload onto it from the web interface such as this:

DELAY 2000
COMMAND SPACE
DELAY 1000
STRING https://www.youtube.com/watch?v=dQw4w9WgXcQ&ab_channel=RickAstley
DELAY 1000
ENTER

It turns into this because of the special characters like slashes and question marks and colons:

DELAY 2000
COMMAND SPACE
DELAY 1000
STRING https%3A%2F%2Fwww.youtube.com%2Fwatch%3Fv%3DdQw4w9WgXcQ%26ab_channel%3DRickAstley
DELAY 1000
ENTER

Is there any fix? This is a rickroll lol

Ojamyjack, 2023-03-18:
It may just be me being stupid but I still couldn’t get it to work maybe I’m inputting data in the wrong place if anybody has an example of the secrets file or if there is a video of someone setting it up could someone give me that link

responsivereality, 2023-03-18:
Sure, I'll send a short video

Ojamyjack, 2023-03-19:
> Sure, I'll send a short video

Where too so I can see it

responsivereality, 2023-03-19:
https://drive.google.com/file/d/1nJIvFESKK31g2-HW-2yvD0DOflEfLeOT/view?usp=drivesdk

It’s a 3 minute video so it’ll take a second to process

Ojamyjack, 2023-03-19:
Thanks that works just gotta find out how to run commands I can write new ones but no button to run

rickviper, 2023-03-21:
I have set up secrets.py by putting ssid and password of my wifi as instructed in the readme section but still my pico w is not connecting to my wifi. Can anyone help me?

responsivereality, 2023-03-21:
The Pico W itself doesn't connect to WiFi. It creates a WiFi network that is by default called "BadAPName", that you then connect to, enter 192.168.4.1 on your browser, and execute payloads wirelessly.

rickviper, 2023-03-21:
Thank u for helping out, it was me being stupid, didn't knew much about how it worked.

closed 2023-05-07 by dbisu